matteocrippa/leafminer #7: Different Versions of GeekMagic Smart TV?

IamGPIO, 2024-01-23:
Had a working GeekMagic Mini TV working with the default firmware, flashed the latest .bin file you have built via the ESP web flasher. Had success flashing and I see the wifi hotspot called LEAFMINER but the screen is black. I have a feeling there are different versions using different screen drivers/protocols. What do you think?

My version is shown below.

Thanks.

(PS, love what you are doing - great work!)

[image]
[image]

matteocrippa, 2024-01-23:
Did you connect to leaf miner WiFi and fill all the data and press save?

Then try to unplug usb and plug it back and see if the screen starts working.

Let me know otherwise we can investigate it!

matteocrippa, 2024-01-23:
Good it works, I will try to improve the setup screen process, so it's more clear what to do next.

closed 2024-01-23 by matteocrippa