matteocrippa/leafminer #8: Reset Wifi Hotspot if bad Wifi Credentials entered

IamGPIO, 2024-01-23:
I had an issues whilst playing last night. I have an obscure wifi password of which can't be seen during the initial setup due to the asterix covering the password text box. 

I stupidly entered the wrong password and it locked the system as presumably, it couldnt connect to the wifi network. I had to erase the ESP8266's memory and reflash.

it would be great if the unit cant find / connect to the specified / input SSID after X seconds it reverts back to it's AP setup mode.

Regards
IamGPIO

matteocrippa, 2024-01-23:
I've just added this feature in v0.0.4, you can see the full changelog in #9 
I tested for now just on a Lilygo-t-s3, feel free to open a new issue if you notice any issue with the SmallTv

closed 2024-01-23 by matteocrippa

IamGPIO, 2024-01-23:
Great stuff, thanks. I shall have a play and keep poking.