matteocrippa/leafminer #17: Can't seem to upload the code

Grantrocks, 2024-03-10:
I cant seem to upload the code. Everything I have tried does nothing. Can there be some kind of instructions for the esp32 lilygo-t-s3?

matteocrippa, 2024-03-10:
@Grantrocks can you provide me any further detail?
Are you uploading using the plaformio and building the source, or uploading using the binary?
Is there any error in the console?

Grantrocks, 2024-03-10:
Sorry I'm uploading via the binary and the esp web tool thing that is linked in the readme.

matteocrippa, 2024-03-10:
@Grantrocks  Please try the FW 0.0.9.

I've just tested with a normal ESP-32-S3 but should work fine for you too.

- I opened https://esp.huhn.me/ using a Chrome based Browser (ARC, Chrome, etc)
- Click the cog and set Baud Speed to 115200
- Attach the ESP-32 via USB
- Click on connect and select my device USB
[image]
- Can happen that device is not listed nor fully connected
[image]
- You need to press the boot button and hit reset
- Later it will connect and you will see something like
[image]
- Setup as in the image and flash
[image]

Grantrocks, 2024-03-11:
Thank you that worked.

closed 2024-03-11 by Grantrocks