matteocrippa/leafminer #20: nodemcu esp8266 v3 not working?

reyxv16, 2024-03-12:
Hello, I am using the nodemcu esp8266 v3 microcontroller and I have not been able to get it to work, is this compatible?
From the arduino console I receive this message when restarting the device:

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v00090940
~ld
rf cal sector: 1020
freq trace enable 0
rf[112] : 0�

In vkbit I don't see the miner working, do I have to wait a certain number of hours to see the results? It seems the code is not working but I can gladly help with the tests.
Thank you very much, excellent work.

reyxv16, 2024-03-13:
> @reyxv16 did you install it via platformio from scratch or using the web installer?

using the web installer and also the esp8266flasher tool, the strange thing is that it seems to be working because it heats up a little but the led does not work (only when starting up)

matteocrippa, 2024-03-13:
@reyxv16 does any WiFi called "LEAFMINER" shows up?
Also did you install all the 3 files using the flash installer or just the firmware?

reyxv16, 2024-03-14:
@matteocrippa Yes, of course, I can see the wifi called LEAFMINER and I can configure the microcontroller, but after restarting it does not seem to work, should the wallet address start with 1, 3 or bc1? I only see a .bin file to install on the esp8266, how can I monitor what the microcontroller is doing?

matteocrippa, 2024-03-14:
The wallet should start with bc1, you can anyhow keep connected the esp8266 to the usb cable and it should show logs if you activate a monitor (eg. using platformio or the web flasher)

reyxv16, 2024-03-16:
[image]
Apparently the software already works, although I still don't see the statistics in vkbit, I'm going to wait a few hours to see if it's working, so far this was the last line received: [I] Network: >>> {"id":3 ,"method":"mining.suggest_difficulty","params":[0.000100]}

cromes85, 2024-03-16:
hello, j'ai le meme, ca commence et puis cela s'arrete...

matteocrippa, 2024-03-16:
Just released v0.0.12, it should fix all the issues

cromes85, 2024-03-16:
Nickel, je vais tester cela

Cependant dans le readme, je pense que vous devriez dire que juste le firmware.bin doit être pousser pour l'esp8266...

Merci pour le job

cromes85, 2024-03-17:
Ça se relance normalement...

reyxv16, 2024-03-17:
[image]
After a few seconds it reboots and connects again...

matteocrippa, 2024-03-17:
It can be the issue with the WDT, I will check, but for now it's not a blocker, it just change the identifier for the mining session due to the reboot

cromes85, 2024-03-17:
Cela devient bloquant car maintenant mes miners ne se reconnectent plus,
Peut etre prevoir un reboot si pas de reponse?

reyxv16, 2024-03-17:
[image]
Apparently it works at times but every few seconds it restarts, it seems that it fails to receive data from the server
14:51:18 [I] Network: <<< [mining.notify] {"id":null,"method": "mining.notify","params":["1caa225","6a712c6effa5e2121be60bffca67794bf0659f8c000127f70000000000000000","020000000100000000000000000000000000000000 000000000000000000000000000000ffffffff150326be0c564b4249542e434f4d","ffffffff033848310000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47 f3378234f260000000016001419a9580b35706e1aca7113ba9088bcb9e4eeb4180000000000000000266a24aa21a9age52e4f3daf1eef2476e6bb79fb00c6e778cf156 8d440c9237c15bcf2c26c50400000000",["1f115725f55c6d90e8b7be9f1da850bedd5b0370f1afe161ee6b99339dfaabc4","39d4d0b3d56ed70e35bd8c2f5ae7460eb451 c1ddaf427ac7a7796fdaa775dcd8","6b3dc13fe83bd40a87f0a8e2a452a87a8c23c7ff6f5cf6a3ddac2cd115d51562","e92ed79d24b8dc12daaf2aff88e7460a5aa54912797ad 596fa688aaac5a3c1bf","2e600417a32d4d4838856ccc378a8ffb8e0783f67158891444a6e7c993105b66","0e5c11073cf6ba67c84fe9770f6425ee7e6dd0fe6e16e2ca97ba dd49136072e1 ","6cfc9ae163b763d665d0c4b5fcf07d38894827ccf45a93a4d78317f2b75c4b14","690c6130f1de2d99da180e16b3a92db9f4bcc25e353bd54649409d8520e92bcc ","57b2732ec12e23ce38b5e5653e5adb01df0da1a80baabdacdcf59be6091cceb8","68cfef8a8e47894b609b93fdc99506fe55c894d37bc67eb193c691b1f1278b5e","c7ffa50395 70ee99ea34c6c73f4c1f699e927e79dbf5d7aebb6a5908925e370d","de54a911462a77d2696834b45e0496e9cc0fb028b170088389f4b19da6d2d2ab"],"20000000","1 7035a59","65f73ba5",false]}

matteocrippa, 2024-03-30:
if it's back working, let's close it!

closed 2024-04-15 by matteocrippa