matteocrippa/leafminer #22: ESP8266 Freezes up on start and never starts mining

wmikrut, 2024-03-16:
Main.cpp on line 95 the code was changed to 
#if defined(ESPS8266)
  network_listen();
#endif

In older versions is was just:

  network_listen();

I removed the #if defined and the module started mining normally.
This was found in 0.0.11

wmikrut, 2024-03-16:
[D] Network: Connecting to host public-pool.io...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.11", null]}

[I] Network: >>> {"id":2,"method":"mining.authorize","params":["BTCAddress.ex1","x"]}

[I] Network: >>> {"id":3,"method":"mining.suggest_difficulty","params":[0.000100]}

[I] Network: <<< [subscribe] {"id":1,"error":null,"result":[[["mining.notify","bafdf3c2"]],"bafdf3c2",4]}
[I] Current: New session id: bafdf3c2
[I] Network: <<< [authorized] {"id":2,"error":null,"result":true}
[I] Network: Authorized
[I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.0001]}
[I] Current: New difficulty: 0.000100000000
[D] Network: Difficulty set to: 0.0001000000
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["18a075f","1f241d23d31cc81385fbca000a3dc118cf115fd00000a1c60000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170343bd0c5075626c69632d506f6f6c","ffffffff021355f325000000001976a91471ff101e5d3e297bfa0eb7ff4db213d81ffb11fb88ac0000000000000000266a24aa21a9edce90035b2b0d70df913894654a493940373e37942b68226386724c14995ac3e900000000",["7704f3daeefdef3aea92193f8350e1be7c42d1893646435c5a5a38c91d41efc5","5f835a574cd4bf8d6270aa2a05fd7f818ec50e6e73c43cda349291b3fbe8e517","8220b56979842db0c5c187d1490d7aa5e02e28cf8212fd2ba8cdaeebf843dd02","7395acaded5967daf6ef11b589348cc101b49db513938dac4545f93c3599b773","3cea7fb12799a52064b3b4ab8bb7985499d2b162aa4b5eb9d78a2efa20af6b8c","91ebfbce1b0fe29bfb1917cb6a18d9a37b86f4a0810e21345ed41dd939e82bae","9ba8933c1e4aa0213dcb89e6bae5f8660f454ec5a0385a05ac20aad5ed87d87f","bf7549e7f6af7abbc0e7aff176645c1b717c5f0ced10f2f04fd92baa915013b9","510b44ff78f44f6a1f196475fbca2acc2bcfabb9496d20e5360f159891fe8d07","5cfd076e9840ce2c6eca24cdab1b772ffe9fdf3325542f58e998062a3a2f674b","d7234e211dd8c18453dcc9871b5776bd0ebb31843daa0076c43590023297493e","581aee6541fcf7643db0c52a9168f1eaf6193df01a968f36857270b387c526cc"],"20000000","17035a59","65f52117",false]}
[I] Job: Random value: 89165024
[I] Job: ExtraNonce2: 89165024
[I] Job: [18a075f] - Coinbase: 02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170343bd0c5075626c69632d506f6f6cbafdf3c289165024ffffffff021355f325000000001976a91471ff101e5d3e297bfa0eb7ff4db213d81ffb11fb88ac0000000000000000266a24aa21a9edce90035b2b0d70df913894654a493940373e37942b68226386724c14995ac3e900000000
[I] Job: [18a075f] - Coinbase hash: c26c23934f32ccb82cdfb726ca0e5f78c9f17b67ed892b26ee3a8d39afc4ddaa
[I] Job: [18a075f] - Merkle Branch [0]: 7704f3daeefdef3aea92193f8350e1be7c42d1893646435c5a5a38c91d41efc5
[I] Job: [18a075f] - Merkle Branch [1]: 5f835a574cd4bf8d6270aa2a05fd7f818ec50e6e73c43cda349291b3fbe8e517
[I] Job: [18a075f] - Merkle Branch [2]: 8220b56979842db0c5c187d1490d7aa5e02e28cf8212fd2ba8cdaeebf843dd02
[I] Job: [18a075f] - Merkle Branch [3]: 7395acaded5967daf6ef11b589348cc101b49db513938dac4545f93c3599b773
[I] Job: [18a075f] - Merkle Branch [4]: 3cea7fb12799a52064b3b4ab8bb7985499d2b162aa4b5eb9d78a2efa20af6b8c
[I] Job: [18a075f] - Merkle Branch [5]: 91ebfbce1b0fe29bfb1917cb6a18d9a37b86f4a0810e21345ed41dd939e82bae
[I] Job: [18a075f] - Merkle Branch [6]: 9ba8933c1e4aa0213dcb89e6bae5f8660f454ec5a0385a05ac20aad5ed87d87f
[I] Job: [18a075f] - Merkle Branch [7]: bf7549e7f6af7abbc0e7aff176645c1b717c5f0ced10f2f04fd92baa915013b9
[I] Job: [18a075f] - Merkle Branch [8]: 510b44ff78f44f6a1f196475fbca2acc2bcfabb9496d20e5360f159891fe8d07
[I] Job: [18a075f] - Merkle Branch [9]: 5cfd076e9840ce2c6eca24cdab1b772ffe9fdf3325542f58e998062a3a2f674b
[I] Job: [18a075f] - Merkle Branch [10]: d7234e211dd8c18453dcc9871b5776bd0ebb31843daa0076c43590023297493e
[I] Job: [18a075f] - Merkle Branch [11]: 581aee6541fcf7643db0c52a9168f1eaf6193df01a968f36857270b387c526cc
[I] Current: Job: 18a075f ready to be mined
[D] Current: Hashrate: 21.01 kH/s
[D] Current: Hashrate: 21.01 kH/s
[I] Miner: [0] > [18a075f] > 0x0003676c - diff 0.018183127478
[I] Network: >>> {"id":5,"method":"mining.submit","params":["BTCAddress.ex1","18a075f","89165024","65f52117","0003676c"]}

[I] Current: New highest hashed difficulty: 0.018183127478
[D] Current: Hashrate: 11.82 kH/s
[D] Current: Hashrate: 20.99 kH/s
[D] Current: Hashrate: 21.00 kH/s
[I] Miner: [0] > [18a075f] > 0x00061c4f - diff 0.000520113152
[I] Network: <<< [mining.submit] {"id":5,"error":null,"result":true}
[I] Network: Share accepted
[I] Current: Hash accepted: 1
[I] Network: >>> {"id":6,"method":"mining.submit","params":["BTCAddress.ex1","18a075f","89165024","65f52117","00061c4f"]}

[D] Current: Hashrate: 13.12 kH/s
[D] Current: Hashrate: 21.01 kH/s
[D] Current: Hashrate: 21.02 kH/s
[I] Miner: [0] > [18a075f] > 0x000caae1 - diff 0.000145842109
[I] Network: <<< [mining.submit] {"id":6,"error":null,"result":true}
[I] Network: Share accepted
[I] Current: Hash accepted: 2
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["18a0bf8","1f241d23d31cc81385fbca000a3dc118cf115fd00000a1c60000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170343bd0c5075626c69632d506f6f6c","ffffffff021355f325000000001976a91471ff101e5d3e297bfa0eb7ff4db213d81ffb11fb88ac0000000000000000266a24aa21a9edce90035b2b0d70df913894654a493940373e37942b68226386724c14995ac3e900000000",["7704f3daeefdef3aea92193f8350e1be7c42d1893646435c5a5a38c91d41efc5","5f835a574cd4bf8d6270aa2a05fd7f818ec50e6e73c43cda349291b3fbe8e517","8220b56979842db0c5c187d1490d7aa5e02e2 [... cut ...]
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: sys
sp: 3ffff560 end: 3fffffb0 offset: 0010
3ffff570:  00000000 0ccccccc 3fff1070 40229675  
3ffff580:  000007ff 3ffff5e8 3fff1070 40226355  
3ffff590:  0000028f 00000020 00000000 40226459  
3ffff5a0:  3fff1070 00000020 3fff2f7c 3fff9f4c  
3ffff5b0:  3fff9f64 00000020 3fff2f7c 3fff9f4c  
3ffff5c0:  3fff9f64 3fff9f7c 000004f5 4021550a  
3ffff5d0:  3fff2f7c 00000000 000004f5 4010047c  
3ffff5e0:  40103ad8 00080000 6469227b 756e3a22

matteocrippa, 2024-03-16:
Seems it somehow takes too long and the watchdog reset the device.
Try the v.0.0.12 I force the watchdog to be fed, so it should not cause a reset

cromes85, 2024-03-17:
even in version 12 it reboots every 30 seconds and I'm not sure that the servers accept...

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3424, room 16 

tail 0

chksum 0x2e

load 0x3fff20b8, len 40, room 8 

tail 0

chksum 0x2b

csum 0x2b

v0009edb0

~ld

��n�r��n|�l�l`bbrl�nb�nl`�rl�l��[I] Main: LeafMiner - v.0.0.12 - (C: 1)

[I] Main: ESP8266 - Disable WDT

[I] Storage: Setup: OK

[I] Blink: Setup

[I] Blink: Blinking 5 times

[I] AutoUpdate: Connecting to NOVA_INV...

[D] AutoUpdate: payload: {

  "current": "0.0.12",

  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",

  "devices": [

    "esp8266",

    "esp32",

    "esp32-s2",

    "esp32-s3",

    "geekmagic-smalltv",

    "lilygo-t-display-s3"

  ]

}

[D] AutoUpdate: Remote Version: 0.0.12

[D] AutoUpdate: No Updates, Version: 0.0.12

[I] Network: Connecting to NOVA_INV...

[D] Network: Connecting to host public-pool.io...

[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.12", null]}

[I] Network: >>> {"id":2,"method":"mining.authorize","params":["3EURCPqy66K2BuK3CdzNJDhm2vGwAp5X8p.1",""]}

[... 460 lines cut ...]
chksum 0x2e

load 0x3fff20b8, len 40, room 8 

tail 0

chksum 0x2b

csum 0x2b

v0009edb0

~ld

��n�r��n|�l�l`bbrl�nB�nl`�rl�l��[I] Main: LeafMiner - v.0.0.12 - (C: 1)

[I] Main: ESP8266 - Disable WDT

[I] Storage: Setup: OK

[I] Blink: Setup

[I] Blink: Blinking 5 times

[I] AutoUpdate: Connecting to NOVA_INV...

cromes85, 2024-03-17:
currently esp8266, reboot by cycle as if it lost the connection, or I don't really know....

=>

[I] Main: ESP8266 - Disable WDT

[I] Storage: Setup: OK

[I] Blink: Setup

[I] Blink: Blinking 5 times

[I] AutoUpdate: Connecting to NOVA_INV...

[D] AutoUpdate: payload: {

  "current": "0.0.12",

  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",

  "devices": [

    "esp8266",

    "esp32",

    "esp32-s2",

    "esp32-s3",

    "geekmagic-smalltv",

    "lilygo-t-display-s3"

  ]

}

[D] AutoUpdate: Remote Version: 0.0.12

[D] AutoUpdate: No Updates, Version: 0.0.12

[I] Network: Connecting to NOVA_INV...

[D] Network: Connecting to host public-pool.io...

[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.12", null]}

[I] Network: >>> {"id":2,"method":"mining.authorize","params":["3EURCPqy66K2BuK3CdzNJDhm2vGwAp5X8p.1",""]}

[I] Network: >>> {"id":3,"method":"mining.suggest_difficulty","params":[0.000100]}

[I] Network: <<< [subscribe] {"id":1,"error":null,"result":[[["mining.notify","59d291c7"]],"59d291c7",4]}

[I] Current: New session id: 59d291c7

[I] Network: <<< [authorized] {"id":2,"error":null,"result":true}

[I] Network: Authorized

[I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.0001]}

[I] Current: New difficulty: 0.000100000000

[D] Network: Difficulty set to: 0.0001000000

[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1b626e2","07b75bc4e1bea563a32a11671c3ef104d82d487e000191000000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170343be0c5075626c69632d506f6f6c","ffffffff022e05c9250000000017a9148c37a7be90c9471bfdf7230420f54c40aa6d7cc1870000000000000000266a24aa21a9ed852fa05be312d1e337e575c4776ab9e90c606bf0a6a556cb7d47f032903c5ebb00000000",["4f442818573f545e40031f9d7468eafa3fce47e9d6211def408f97ddb05d15c2","b3a4d9a68b931323d201d2b9214d03057717467f7a575267c39e854c3d050bb4","72bd74d98f6dfc3244e6e8b013215f94984d4877c1474f87eb9873fba83446dd","931dc0d6b467efa8826ac1861176ccc1f44f1ecaeee5ec07ea1ec2759239c857","a24636ddfa26be49140d2e7b57b79e28909e662a23c9526156d5f5ba247b41cd","1f45f42a5b98f2f471cf862b69b92f5d7e330f63b0cea12f1b6cee6735422d64","95a9aae33c3c77dbe03e7895006a0e1dbbcef5b4714461ac9dabc66b9276096f","496b127db384982fd4681b5bcfbecedade2dfb199ac3674b860174ea8be380d6","1afcf4cc4f15a3636babb568ca4f6c17c1ccda13a0bc6360e0d45f54688f25d9","fc78e4798c8cef0c671108d0778177a36a5944efa7b68da8d88c94afd1cb4245","f5d112847d5cdbe62095451285f939dae651f612feb570f5c86491e7c56802b6","bf38bc4f84bc22bb27250f4c3df0e0b9d50f09bc532d1b7f59c81606abf97637"],"20000000","17035a59","65f77a97",false]}

[I] Job: Random value: 1848596091

[... 454 lines cut ...]
chksum 0x2e

load 0x3fff20b8, len 40, room 8 

tail 0

chksum 0x2b

csum 0x2b

v0009edb0

~ld

��n�r��n|�l�l`bbrl�nb�nl`�rl�l��[I] Main: LeafMiner - v.0.0.12 - (C: 1)

[I] Main: ESP8266 - Disable WDT

[I] Storage: Setup: OK

[I] Blink: Setup

[I] Blink: Blinking 5 times

[I] AutoUpdate: Connecting to NOVA_INV...

cromes85, 2024-03-17:
[I] Current: Job: 1ba8ed1 ready to be mined

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.87 kH/s

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.86 kH/s

[I] Miner: [0] > [1ba8ed1] > 0x00078534 - diff 0.000155613845

[I] Network: >>> {"id":5,"method":"mining.submit","params":["3EURCPqy66K2BuK3CdzNJDhm2vGwAp5X8p.1","1ba8ed1","11778679","65f77c96","00078534"]}

[I] Network: <<< [mining.submit] {"id":5,"error":null,"result":true}

[I] Network: Share accepted

[I] Current: Hash accepted: 1

[I] Current: New highest hashed difficulty: 0.000155613845

[D] Current: Hashrate: 10.62 kH/s

[D] Current: Hashrate: 16.87 kH/s

[I] Miner: [0] > [1ba8ed1] > 0x00085901 - diff 0.000187118598

[I] Network: >>> {"id":6,"method":"mining.submit","params":["3EURCPqy66K2BuK3CdzNJDhm2vGwAp5X8p.1","1ba8ed1","11778679","65f77c96","00085901"]}

[I] Network: <<< [mining.submit] {"id":6,"error":null,"result":true}

[I] Network: Share accepted

[I] Current: Hash accepted: 2

[I] Current: New highest hashed difficulty: 0.000187118598

[D] Current: Hashrate: 4.87 kH/s

[D] Current: Hashrate: 16.85 kH/s

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.84 kH/s

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.87 kH/s

[D] Current: Hashrate: 16.86 kH/s

[D] Current: Hashrate: 16.86 kH/s

[I] Miner: [0] > [1ba8ed1] > 0x0010bb85 - diff 0.000451916177

[I] Network: >>> {"id":7,"method":"mining.submit","params":["3EURCPqy66K2BuK3CdzNJDhm2vGwAp5X8p.1","1ba8ed1","11778679","65f77c96","0010bb85"]}

[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1ba93a8","ab1331548dc58dd8dd45253efe09d9922e7eb5fe00034fce0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170344be0c5075626c69632d506f6f6c","ffffffff02d543bc250000000017a9148c37a7be90c9471bfdf7230420f54c40aa6d7cc1870000000000000000266a24aa21a9edf7fb3995f2b869b14510a6b73a0d3dcdcb079cf0bc02a00299f508076799530d00000000",["caa3d344cd7ea131153423e201968cb1909924308c4dcd7a8c0caab0d2d65085","b85b8cfc13e7656c108d64f825aa0c5dfc0a3a0757df0dcaa0752607f7c7aa02","8b27fcab1f0e7192e2069805be63d8be73ea6c996ed2fff65b497f8fe8c4882c","61f3aa39cceacef1114188e1f4e4fac9fbd06fe1604c979bb6a484e67dd212c2","5735a97be86aad71ae67b1f07b0ad3f03ada4ecb659a4b8025cd74637dc4fe6d","5152c899aa88ab116eb1c9ad1104e4a036f29c994f45e2652e40f5f4ad5319b6","e7a6da2c539f78bc37db8c53017b10c08e00d70d173d778361b2828246f26d50","9678fe97b62bc2377a1be48ef126f967ed5388e1a769084bc68627b4ee99c985","3701491da1e840d3e186cd14acc417ce915d6e1a3f1d6a7fbcd669e0bad39196","6b1ec2cd8a6fd3d711578030cd3847eca2bed7383df10638a526e8adc35927cb","dd465c2041ba6e12004bc319fd5d23735241eb55d1514c1a504b5e700e3f6f92","4b11cbfc79079610af2e50b238e3d3a8d9c279ce8bafda6bff7a27ab9a5eb01d"],"20000000","17035a59","65f77cd2",false]}

[I] Job: Random value: 2050646087

[I] Job: ExtraNonce2: 20506460

[I] Job: [1ba93a8] - Coinbase: 02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170344be0c5075626c69632d506f6f6c938e5cdf20506460ffffffff02d543bc250000000017a9148c37a7be90c9471bfdf7230420f54c40aa6d7cc1870000000000000000266a24aa21a9edf7fb3995f2b869b14510a6b73a0d3dcdcb079cf0bc02a00299f508076799530d00000000

[I] Job: [1ba93a8] - Coinbase hash: ff9d0e3014b4cc436db4c1a592f6c606d97f62e1791cf66fb11c406c20bfea83

[... 342 lines cut ...]
3ffffef0:  3fffff20 3fffff00 00000008 27a2c281  

3fffff00:  48bc9a7f 00492c83 00000000 3fffa09c  

3fffff10:  0010bb85 0027ab07 3fffff54 3fff12dc  

3fffff20:  00000000 3ef384ed e56ed912 3fff0dd4  

3fffff30:  00000000 3f3d9de5 32e55a91 40207a12  

3fffff40:  32e55a91 3f3d9de5 3fff1010 3fff12dc  

3fffff50:  0010bb85 11f32c50 d2cedf71 0d2ee781  

3fffff60:  192c83f6 067ba686 8a0d98ad c417415b  

3fffff70:  000008a4 3fff0ddc 00000000 4021533b  

3fffff80:  3ffeab24 00000000 00000001 3fff12dc  

3fffff90:  3fffdad0 00000000 3fffdab0 40207595  

3fffffa0:  feefeffe feefeffe 3fffdab0 40229612  

<<<stack<<<

wmikrut, 2024-03-18:
I loaded up version 12 compiled, uploaded and monitored with Platform IO.
I am also running a local stratum proxy to see what happens with the submitted shares.  I am definitely seeing shares submitted, so no issue there.

However, I do see kernel panics at different intervals.
Here are some logs with the decoder output.

[esp8266.log](https://github.com/matteocrippa/leafminer/files/14629693/esp8266.log)

wmikrut, 2024-03-18:
Here is 10 minutes of run time from PlatformIO serial monitor with decoder.

Could this be an issue with parsing a network response?
Each time it fails it seems to be on network_listen() at src\network/network.cpp:416

[esp8266.log](https://github.com/matteocrippa/leafminer/files/14629800/esp8266.log)

wmikrut, 2024-03-18:
The original reason for this issue has been resolved.
I'll move the crash to a new issues thread.

closed 2024-04-19 by matteocrippa