matteocrippa/leafminer #23: Impossible Compil for on vsc and PlatFormIO

cromes85, 2024-03-17:
**Describe the bug**
Impossible de compiler

**To Reproduce**
Steps to reproduce the behavior:

Processing esp8266 (platform: espressif8266; board: d1_mini; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.0.1) > WeMos D1 R2 and mini
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2) 
 - tool-esptool @ 1.413.0 (4.13) 
 - tool-esptoolpy @ 1.30000.201119 (3.0.0) 
 - tool-mklittlefs @ 1.203.210628 (2.3)
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 40 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TridentTD_ESP_TrueRandom @ 1.0.0+sha.0ff7623
|-- cJSON @ 0.0.0+20240317212805.sha.87d8f09
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4
|-- Preferences @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
Building in debug mode
Compiling .pio\build\esp8266\src\network\accesspoint.cpp.o
Archiving .pio\build\esp8266\lib13a\libESPAsyncTCP.a
Archiving .pio\build\esp8266\lib0f8\libHash.a
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiGeneric.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiGratuitous.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiMulti.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiSTA-WPS.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiSTA.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\ESP8266WiFiScan.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\WiFiClient.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\WiFiClientSecureBearSSL.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\WiFiServer.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\WiFiServerSecureBearSSL.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\WiFiUdp.cpp.o
Compiling .pio\build\esp8266\lib26f\ESP8266WiFi\enable_wifi_at_boot_time.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\AsyncEventSource.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\AsyncWebSocket.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\SPIFFSEditor.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\WebAuthentication.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\WebHandlers.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\WebRequest.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\WebResponses.cpp.o
Compiling .pio\build\esp8266\liba12\ESP Async WebServer\WebServer.cpp.o
Compiling .pio\build\esp8266\lib560\LittleFS\LittleFS.cpp.o
Compiling .pio\build\esp8266\lib560\LittleFS\lfs.c.o
src\network\accesspoint.cpp: In function 'void accesspoint_print_error()':
src\network\accesspoint.cpp:62:32: error: 'class UpdaterClass' has no member named 'getErrorString'
   62 |     std::string error = Update.getErrorString().c_str();
      |                                ^~~~~~~~~~~~~~
In file included from .pio\libdeps\esp8266\ESP Async WebServer\src\SPIFFSEditor.cpp:1:
.pio\libdeps\esp8266\ESP Async WebServer\src\SPIFFSEditor.h:16:101: warning: 'SPIFFS' is deprecated: SPIFFS has been deprecated. Please consider moving to LittleFS or 
other filesystems. [-Wdeprecated-declarations]                                                                                                                            16 |     SPIFFSEditor(const String& username=String(), const String& password=String(), const fs::FS& fs=SPIFFS);
      |                                                                                                     ^~~~~~
In file included from .pio\libdeps\esp8266\ESP Async WebServer\src/ESPAsyncWebServer.h:27,
                 from .pio\libdeps\esp8266\ESP Async WebServer\src\SPIFFSEditor.h:3,
                 from .pio\libdeps\esp8266\ESP Async WebServer\src\SPIFFSEditor.cpp:1:
C:\Users\Home-Pc1\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:275:15: note: declared here
  275 | extern fs::FS SPIFFS __attribute__((deprecated("SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems.")));      
      |               ^~~~~~
Compiling .pio\build\esp8266\lib560\LittleFS\lfs_util.c.o
Compiling .pio\build\esp8266\lib09c\Preferences\Preferences.cpp.o
*** [.pio\build\esp8266\src\network\accesspoint.cpp.o] Error 1
============================================================= [FAILED] Took 6.73 seconds =============================================================

 *  Arrêt du processus de terminal "C:\Users\Home-Pc1\.platformio\penv\Scripts\platformio.exe 'run', '--target', 'upload'". Code de sortie : 1.

matteocrippa, 2024-03-17:
Checking at the code it seems you are using some different version of the libraries.
I just run it locally and in the Github CI and it worked fine.

I see that the most relevant differences are:

`PLATFORM: Espressif 8266 (4.2.1) > WeMos D1 R2 and mini`

In your log seems you are using an older version: PLATFORM: Espressif 8266 (**4.0.1**) > WeMos D1 R2 and mini

Same for the packages
```
PACKAGES: 
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2) 
```

vs

```
framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
```

Also the error that breaks the compilation is this one:

`src\network\accesspoint.cpp:62:32: error: 'class UpdaterClass' has no member named 'getErrorString'
62 | std::string error = Update.getErrorString().c_str();`

It seems that Update.h class has not getErrorString() function, can be for the issue above of the different version.

I will send later a commit with 0.0.13 trying to force to use the latest version of the libs

cromes85, 2024-03-17:
Do you have a version of the code that can be used in arduinoIDE, I am more comfortable and I would like to help evolve?

matteocrippa, 2024-03-30:
@cromes85 did you try now using the main branch?

cromes85, 2024-03-31:
Pas encore

matteocrippa, 2024-04-19:
@cromes85 still same error?