matteocrippa/leafminer #24: ESP8266 Crashes at seemingly random intervals after mining starts

wmikrut, 2024-03-18:
After a short period of time the ESP8266 crashes and restarts.
Mine on an pool and wait a few minutes to reproduce the issue.

I see that shares are being submitted and can verify that those sahes are making it through my local stratum proxy to the pool.

Here is 10 minutes of run time from PlatformIO serial monitor with decoder.
I am running a local stratum proxy and I definitely see valid shares being submitted to the pool.

Could this be an issue with parsing a network response?
Each time it fails it seems to be on network_listen() at src\network/network.cpp:416

I have experienced this with v11 and v12.

[esp8266.log](https://github.com/matteocrippa/leafminer/files/14630531/esp8266.log)

cromes85, 2024-03-19:
@matteocrippa 
Avez vous une possibilité pour cela?

matteocrippa, 2024-03-19:
nope, will try during a night this week

cromes85, 2024-03-19:
Rien d'urgent pour moi... je voudrais pouvoir aider...
Mais mon domaine est plus le python...

wmikrut, 2024-03-20:
I know why the crash is occurring.

I ran the miner with a stratum proxy and was watching the screen.
Whenever the proxy notified about a new job available:

2024-03-20 14:03:41,301 INFO proxy client_service.handle_event # New job 201819d for prevhash 78501535, clean_jobs=False

The miner crashes and restarts.  
This is why it seems random, new notifications come all the time.

cromes85, 2024-03-20:
@wmikrut 
comment pensez vous que l'on pourrait résoudre cela?
how do you think we could resolve this?

matteocrippa, 2024-03-20:
Good catch, so probably is going out of memory when it calculates the coinbase

cromes85, 2024-03-21:
```
#ifndef UNIT_TEST

#include <Arduino.h>

#if defined(ESP32)
#include "freertos/task.h"
#endif // ESP32

#include "leafminer.h"
#include "utils/log.h"
#include "model/configuration.h"
#include "network/network.h"
#include "network/accesspoint.h"
#include "utils/blink.h"
#include "miner/miner.h"
#include "current.h"
#include "utils/button.h"
#include "storage/storage.h"
#include "network/autoupdate.h"
#include "massdeploy.h"
#if defined(HAS_LCD)
#include "screen/screen.h"
#endif // HAS_LCD

char TAG_MAIN[] = "Main";
Configuration configuration = Configuration();
bool force_ap = false;

void setup()
{
  Serial.begin(115200);
  delay(3000); // augmentation pour test
  l_info(TAG_MAIN, "LeafMiner - v.%s - (C: %d)", _VERSION, CORE);

#if defined(ESP8266)
  l_info(TAG_MAIN, "ESP8266 - Disable WDT");
  ESP.wdtDisable();
  *((volatile uint32_t *)0x60000900) &= ~(1);
#else
  l_info(TAG_MAIN, "ESP32 - Disable WDT");
  disableCore0WDT();
#endif // ESP8266

  storage_setup();

  force_ap = button_setup();

  storage_load(&configuration);

  if (configuration.wifi_ssid == "" || force_ap)
  {
#if defined(MASS_WIFI_SSID) && defined(MASS_WIFI_PASS) && defined(MASS_POOL_URL) && defined(MASS_POOL_PASSWORD) && defined(MASS_POOL_PORT) && defined(MASS_WALLET)
    configuration.wifi_ssid = MASS_WIFI_SSID;
    configuration.wifi_password = MASS_WIFI_PASS;
    configuration.pool_url = MASS_POOL_URL;
    configuration.pool_password = MASS_POOL_PASSWORD;
    configuration.pool_port = MASS_POOL_PORT;
    configuration.wallet_address = MASS_WALLET;
#else
    accesspoint_setup();
    return;
#endif // MASS_WIFI_SSID && MASS_WIFI_PASS && MASS_SERVER_DOMAIN && MASS_SERVER_PASSWORD && MASS_WALLET
  }

#if !defined(HAS_LCD)
  Blink::getInstance().setup();
  delay(500);
  Blink::getInstance().blink(BLINK_START);
#else
  screen_setup();
#endif // HAS_LCD

  autoupdate();

  if (network_getJob() == -1)
  {
    l_error(TAG_MAIN, "Failed to connect to network");
    l_info(TAG_MAIN, "Fallback to AP mode");
    force_ap = true;
    accesspoint_setup();
    return;
  }

#if defined(ESP32)
  btStop();
  xTaskCreate(currentTaskFunction, "stale", 1024, NULL, 1, NULL);
  xTaskCreate(buttonTaskFunction, "button", 1024, NULL, 2, NULL);
  xTaskCreate(mineTaskFunction, "miner0", 6000, (void *)0, 10, NULL);
  xTaskCreate(networkTaskFunction, "network", 10000, NULL, 3, NULL);
#if CORE == 2
  xTaskCreate(mineTaskFunction, "miner1", 6000, (void *)1, 11, NULL);
#endif
#endif

#if defined(ESP8266)
  network_listen();
#endif
}

void loop()
{
  if (configuration.wifi_ssid == "" || force_ap)
  {
    accesspoint_loop();
    return;
  }

#if defined(ESP8266)
  miner(0);
#endif // ESP8266
}

#endif
```

for test, increase of the delay at line 32 in the main.cpp file

wmikrut, 2024-03-21:
I'm not sure it's an out of memory condition.  I loaded your Job context with breakpoints and the whole process completes every time.  I don't think its a WDT issue because WDT was disabled in main... I re-enabled it and saw the same result.

Could it be as simple as a stack overflow?
[job.cpp.txt](https://github.com/matteocrippa/leafminer/files/14705001/job.cpp.txt)
[esp8266.log](https://github.com/matteocrippa/leafminer/files/14705002/esp8266.log)

cromes85, 2024-03-21:
Et la version 10 fonctionnait?
And version 10 worked?

wmikrut, 2024-03-21:
I compiled all the way back to v5.
There is no v6 Source.

v5 does not crash on new work notifications.
v7 and on does crash on new work notifications.

cromes85, 2024-03-21:
@wmikrut 
and it seems that it updates automatically?

wmikrut, 2024-03-21:
I've been running it for 15 minutes now with no crashes. 
However, there have been importand fixes in later releases.

I'll keep digging and see if I can spot where it's binding up.

cromes85, 2024-03-21:
OK top, Can you send me the functional version in bin without automatic update in the meantime?  in *.bin format?

matteocrippa, 2024-03-21:
Just pushed a few changes, quick tested and seems way more stable for me for a Weimos D1, give a try to 0.0.13 (or just way the autoupdate forcing a reboot)

cromes85, 2024-03-21:
Greeaaattt

wmikrut, 2024-03-22:
It is definitely more stable and now I think I can see a potential memory issue that would be easy to fix.

Every so often the stratum server comes along and assigns a new job and parm 9 will be clean jobs false.
From what I gather it means that the server is telling the miner - "Finish your nonces then start on this new job."
Mainly so you're not waiting for work.

Under normal circumstances this is fine because hardware miners are quick and jump from job to job.

Now the ESP8266 is much slower and can't handle a lot of queued work.  I've see after 3-6 notifications of work being queued the chip freezes up completely.  

I've added some quick code to say when work is already queued up, skip queuing up additional jobs.
Until Clean Jobs True comes down, we can can working on the current job until we run out of nonces and start on the next job.

When true comes down reset everything and start over.

wmikrut, 2024-03-22:
Definitely much more stable with the limiting of queued jobs.
The program is no longer dumping or freezing up.

After an hour it was still running.

It stopped only because by wifi router is garbage and the connection dropped.
This is not an item related to this thread, just an FYI.

The program picked it up and reconnected, but it never re-subscribed or authorized the new connection so I kept getting the Connection is not subscribed error.   A future item -  subscribe(), authorize(), difficulty() on a new connection.

wmikrut, 2024-03-22:
I forked the project so you can look at some of the code I am playing with on my dev branch.

matteocrippa, 2024-03-22:
I prepared a branch `v/0.0.14`, but didn't had time to test it yet with some changes according to the feedback you shared.
In general any 1 core device will skip enqueuing any extra job and just replace the current on `clean_jobs` is `true`.
Also add logic to try to force reset the session if connection is failing

wmikrut, 2024-03-22:
What caught my attention was line 83 in current.cpp
 current_job_next = new Job(notification, *current_subscribe, current_difficulty);

This was firing every time new work was sent down from the proxy with clean jobs = false.
There are times I see a clean jobs false come down 30-40x before a true comes down.

It was only a guess that perhaps the new Job was somehow allocating memory and/or stack space.
I don't know for sure... but I do know what once I stopped the new Job from running on every notification it stopped freezing up.

wmikrut, 2024-03-23:
We can close this issue.
I have moved on to v14 for testing.

wmikrut, 2024-03-23:
I've been running v14 for 30 minutes now.
Not a single error!

21:49:24.498 > [I] Miner: [0] > [247b563] > 0x00188f0f - diff 0.000043553264
21:49:24.498 > [I] Network: >>>
{"id":232,"method":"mining.submit","params":["wmikrut.ex1","247b563","745449","65fe4245","00188f0f"]}
21:49:24.514 >
21:49:24.639 > [I] Network: <<< [mining.submit] {"error": null, "id": 232,
"result": true}
21:49:24.639 > [I] Network: Share accepted
21:49:24.639 > [I] Current: Hash accepted: 200

That's a lot of work for the 8266!
Can't wait to let it run overnight.

matteocrippa, 2024-03-30:
Can we close this?

cromes85, 2024-03-31:
Pas encore svp

PSLLSP, 2024-04-11:
I tried to flash ESP8266MOD with LeafMiner 0.0.13 and it was not stable. It mined for several minutes and then it stopped to mine.
When it was mining and I I was watching serial console then I noticed that miner often connects to WiFi, like it loses signal or it was disconnected.
From my point of view, this issue was not addressed in LeafMiner 0.0.13 and should not be closed. Or there is other issue with ESP8266MOD, it it just not stable miner...

ffrediani, 2024-04-26:
Hello
I have just flashed a ESP8266 v3 (NodeMCU) and I am not sure this issue I am watching. If not related I can open a separate issue.
After flashing it with 0.0.15 and configured it with the SSID, password and wallet address the one thing I've changed was to not show anything on the LCD as this unit had none.

After rebooting it connects and starts mining. I can see shares being submited to vkbit.com, however if I try to simply ping its IP address it responds intermittently, not sure if it is disconnecting and reconnecting the network.
Also I am unable to access the Web interface pointing the browser to its IP address.

matteocrippa, 2024-04-26:
@ffrediani it's correct, web gui is available only at the first boot or if you erase the flash, keeping a web ui up and running will kill the already limited performances of an esp8266.
The behaviour of not being pingable is somehow correct too because we retain a connection, but it's more an open tcp+stratum to the server, as you mentioned you can see it running on vbkit.

If you want there's a branch v0.0.16 with some patches, but it's still a work in progress

ffrediani, 2024-04-26:
@matteocrippa thanks.
Isn't there any way to prioritize the processes (the one runs the web interface over the mining one) when required ? So it could decrease the hashrate as necessary in order to reply to the web client for a short period while it is being accessed ?

In any way, even not pinging it or trying to access the web interface I can see that it disconnects and re-connects to the Wifi Access Point every in a while. Is this expected ? If so, if this is happening while it is solving a block it may not be able to submit a share.

Yeah I can try the 0.0.16, however I was not able to find the .bin in Github and can't compile it myself. Any particular URL I can download from ?

ffrediani, 2024-04-27:
Yeah, seems that after a while mining the ESP8266 disconnects or crashes and doesn't recover anymore until rebooted. Do you think 0.0.16 would have any chance to avoid this behavior ?

matteocrippa, 2024-04-27:
0.0.16 has some patches in that direction but didn't have time to fully test it, for sure it's more stable than 0.0.15 for esp8266. You can install only building manually for now, till it's released via CI.

ffrediani, 2024-04-27:
Well, at the moment there is not much too loose as 0.0.15 on the ESP8266 (NodeMCU) is crashing every in a while and only a reboot brings it back to life. If anyone can make the binary available somewhere I can upgrade here.

ffrediani, 2024-04-27:
@matteocrippa is there any way to prioritize the processes (the one runs the web interface over the mining one) when required ? So it could decrease the hashrate as necessary in order to reply to the web client for a short period while it is being accessed and be able to show the web interface for troubleshooting and upgrade ?

matteocrippa, 2024-04-27:
No plan for such a feature for solo mining, it's already kinda success that an ESP8266 can barely mine.
Introducing something like this risk to be a major pain somehow.
So far the Serial monitor is a good way to troubleshoot and debug, web can be used just for the first setup for non tech users.

ffrediani, 2024-04-27:
It is not a feature. Feature is an extra or new thing, but prioritizing processes, if possible for ESP8266, is already something that is already there.

I managed to compile 0.0.16 in PlatformIO and upload it. I am monitoring on the serial monitor. If I find anything before the crash I will share.
Thanks

matteocrippa, 2024-04-27:
0.0.16 is available as binary too if you need.
Sadly turns into a feature because it will require some logic rewriting to achieve something that can introduce lower performances, on ESP8266 in particular I was trying to strip out all the possible extra to have the best performance due to the limited cpu power.

ffrediani, 2024-04-29:
Hi

I have been using 0.0.16 for about 2 days. I left it running connected to the PC with a Serial Monitor. It had some issues about 2 times, but it recovers by rebooting and starting over again so the mining process doesn't die and all stops.
When I connected it to a simple USB power supply to the wall, it starts mining but after a while it dies and as it is connected to the USB power supply it is not possible to watch the serial.

How the fact to be connected to the serial monitor can change its behavior in a way that it didn't die one when connected to it ? Any particular debug/troubleshooting or data collection I can make in order to find out more?

PSLLSP, 2024-04-29:
> How the fact to be connected to the serial monitor can change its behavior in a way that it didn't die one when connected to it ?

Maybe serial console, that TX buffer is not read by PC and buffer overflows?

ffrediani, 2024-04-29:
Hi. I've connected again to serial console to watch it and left it running for a while. After some random time it simply stopped mining but didn't throw any errors. Find below the last couple of messages I see in the serial console. Let me know if there is anything else I can collect or do in order to get more information that identifies the problem.

[D] Network: <<< len: 37
[I] Network: <<< [mining.submit] {"id":275,"error":null,"result":true}
[I] Network: Share accepted
[I] Current: Hash accepted: 234
[D] Network: <<< len: 0
[D] Miner: [0] > Hash 0.000052475481 > 0.000016000000
[I] Miner: [0] > [6211ad] > 0x004becb1 - diff 0.000052475481
[I] Network: >>> {"id":277,"method":"mining.submit","params":["<wallet-hidden-address>.ESP8266-NodeMCU","6211ad","11221363","662fcbc0","004becb1"]}

[D] Network: <<< len: 37
[I] Network: <<< [mining.submit] {"id":277,"error":null,"result":true}
[I] Network: Share accepted
[I] Current: Hash accepted: 235
[D] Network: <<< len: 721
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["6242b1","a3afdcbb1af9ab21edb6f35751204f96dda3ab240001667e0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1503abd60c564b4249542e434f4d","ffffffff025074a31200000000160014e55dea92fb04b1d68e692a016f29989966d538eb0000000000000000266a24aa21a9ed461f6b6de58a3aec52ec7d1e9fb573fea92fe609fe87cd350de0b8a9dec285b800000000",["dd22d0b2a0243176f7f0eef6f8ee570f53d76a16c57caa8cc37371123c4a67d7","766881d6e824159d63f7d43613e12c0611329d06e22fd461a6504b0c292594c7","cd4b33e458e3318cdbb5d4ad7a140ff286072481620b6622713deda12f7392c7","e65fa7836e99d150e7c35ded5f47a975437204cd44395c85c83ddd32e843df04"],"20000000","170331db","662fcddb",true]}

[D] Current: Job: 6211ad is cleaned and replaced with 6242b1
[I] Job: Random value: 1044545964

ffrediani, 2024-04-29:
To add up managed to catch up a different error on the serial console. On this time however it "recover" by starting over again on reboot:

[I] Job: Random value: 1964889275
[I] Current: Job: 62483c ready to be mined
[D] Network: <<< len: 54
[I] Network: <<< [mining.submit.fail] {"id":8,"result":null,"error":[21,"Job not found",""]}
[E] Network: Share rejected
[E] Network: Late responses, skip them
[D] Network: <<< len: 63
[I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.00001]}
[I] Current: New difficulty: 0.000010000000
[D] Network: Difficulty set to: 0.0000100000
[D] Network: <<< len: 783

[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["624857","ce1c8a9a80ef4ed23a53a2fb74117cd532dbc4ed000164fa0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1503afd60c564b4249542e434f4d","ffffffff038fdd17000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3395398d1200000000160014e55dea92fb04b1d68e692a016f29989966d538eb0000000000000000266a24aa21a9ed181cd5416c2d26099b6873b2514ef6ae3d0d9e0438f2a4bec2d813958d05fdaa00000000",["eb85016e05b7149817abc553d40fb18f3f7a1fc547a07ef9f015253d53ca36b5","ab96f6c6e5a76f22b77a1e58dc4a28aa381fa727b21f60ac02b29b2a8a50e079","ec5bc63184fef71824c363b89ab350e63cfd0b43e60b0fa0b09947a96323044b","fba15909307f25533d3f723f4fa8463bd30c21684e9da0ed014a8badcae97744"],"20000000","170331db","662fdb00",true]}

[D] Current: Job: 62483c is cleaned and replaced with 624857
[I] Job: Random value: owork: <<< len: 54
[I] Network: <<< [mining.submit.fail] {"id":9,"result":null,"error":[21,"Job not found",""]}
[E] Network: Share rejected
[E] Network: Late responses, skip them
[D]  00000000 40104919 3ffefb30  

3ffffa40:  0000003c 00000000 40104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
3ffffa50:  0000001c 00000001 40105a61 3ffef104919 000000ff  
(several lines like the ones above) 
...
...
3ffffd10:  3fffffd00:  3fff363c 00000099 3fff36ad 00000000  

3ffffd10:  3f 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0009f7e0

~ld

��n�r��n|�l�l`bbrl�nb�nl`�rl�l��[I] Main: LeafMiner - v.0.0.16 - (C: 1)

[I] Main: Compiled: Apr 27 2024 17:14:19
[I] Main: Free memory: 36248
[I] Main: ESP8266 - Disable WDT
[I] Storage: Setup: OK
[I] Configuration: wifi_ssid: MySSID
[I] Configuration: wifi_password: mypassword
[I] Configuration: wallet_address: wallet-address.ESP8266-NodeMCU

[I] Configuration: pool_password: x
[I] Configuration: pool_url: pool.vkbit.com
[I] Configuration: pool_port: 1073741680
[I] Configuration: blink_enabled: on
[I] Configuration: blink_brightness: 256
[I] Configuration: lcd_on_start: off
[I] Configuration: miner_type: 
[I] Configuration: auto_update: on
[I] Blink: Setup
[I] Blink: Blinking 5 times

[I] AutoUpdate: Connecting to MySSID...
[D] AutoUpdate: payload: {
  "current": "0.0.16",
  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",
  "devices": [
    "esp8266",
    "esp32",
    "esp32-s2",
    "esp32-s3",
    "geekmagic-smalltv",
    "lilygo-t-display-s3"
  ]
}

[D] AutoUpdate: Remote Version: 0.0.16
[D] AutoUpdate: No Updates, Version: 0.0.16
[I] Network: Connecting to MySSID...

ffrediani, 2024-04-29:
Managed to catch another error which also recovered by a reboot

[I] Current: Hash accepted: 821
[D] Network: <<< len: 0
[D] Miner: [0] > Hash 0.000074538038 > 0.000016000000
[I] Miner: [0] > [663d05] > 0x0036c176 - diff 0.000074538038
[I] Network: >>> {"id":954,"method":"mining.submit","params":["wallet-address.ESP8266-NodeMCU","663d05","66321160","662ff998","0036c176"]}

[D] Network: <<< len: 37
[I] Network: <<< [mining.submit] {"id":954,"error":null,"result":true}
[I] Network: Share accepted
[I] Current: Hash accepted: 822
[D] Network: <<< len: 721
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["6662c1","78a55fdce045fcec0a8ebf1027d9eee88af815010000c74e0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1503bed60c564b4249542e434f4d","ffffffff02fe92a31200000000160014e55dea92fb04b1d68e692a016f29989966d538eb0000000000000000266a24aa21a9ed4f526208d9c69ee7153714537088d8f99037e3d423eedded7e05560147a5fdd100000000",["65e63f9daece411de58adb9543f2db626f6aead247b459718a016a1082b9d5b6","9ac7ed2108745edfff66d944f1a7c5871c25429aeceff11228c4104735fefac6","07c0f640c728a68e394bd60085a253cf935bd6566d1b6b7132cfc41bc12f3035","5ee51d4725fd666e4616691947cfe3ee7e036abbe3af82fdcb185318ad89d9b2"],"20000000","170331db","662ffb20",true]}

[D] Current: Job: 663d05 is cleaned and replaced with 6662c1
[I] Job: Random value: 267434342
[I] Current: Job: 6662c1 ready to be mined

User exception (panic/abort/assert)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield
>>>stack>>>
ctx: cont
sp: 3ffff560 end: 3fffffc0 offset: 0010
3ffff570:  00000001 62d0e560 3fff12d0 40229b7d  
3ffff580:  000007ff 3ffff5e8 3fff12d0 4022685d  
3ffff590:  00000527 00000020 00000000 40226961  
3ffff5a0:  3fff12d0 00000020 3fff469c 3fff45dc  
3ffff5b0:  3fff45f4 00000020 3fff469c 3fff45dc  
3ffff5c0:  3fff45f4 3fff460c 000002d1 402157fe
...
...
3ffffce0:  0000009b 3ffe8368 0000009b 40237901  
3ffffcf0:  00000000 00000000 00000001 4010047c  
3ffffd00:  3fff363c 0000009a 3fff36ac 00000000  
3ffffd10:  3fff36ad��n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l����n�r��n|�l�l`bbrl�nb�nl`�rl�l��_brightness: 256

[I] Configuration: lcd_on_start: off
[I] Configuration: miner_type: 
[I] Configuration: auto_update: on
[I] Blink: Setup
[I] Blink: Blinking 5 times

matteocrippa, 2024-04-29:
Seems mostly related to the change of the job, btw good that it reboots and does not stay stuck
It would be perfect to decode the crash https://github.com/dankeboy36/esp-exception-decoder

ffrediani, 2024-04-29:
@matteocrippa yeah, most of the time it reboots which is Ok to keep going, however one of the reports above it simply stops, doesn't show any errors and only recovery when a Poweroff/Poweron. I just got back to the PC and again it was stopped without any errors. Just the last message "Job: Random value: 62289229" and that's it.

Strangely it happens mostly (but not only) when it is connected directly to a power supply where is not possible to get serial output.

ffrediani, 2024-05-01:
Hello @matteocrippa 
I see that on the changelog of comming 0.0.17 version there is nothing else other than the version change. Did you, by any chance, find anything relavant that was causing it to stop mining after random times after boot by?

I have been running 0.0.16 here on the serial console and most of the time it just gets stuck with no error shown, stops mining wihtout a reboot or anything else and just a poweroff/poweron reovers it.
I was going to suggest maybe adding a watchdog to check if it stoppped mining and reboot or restart the job, if that make sense for now.

Anyway, I have flashed 0.0.17 already to the ESP8266 and I'm testing it. Will provide feedback soon.

matteocrippa, 2024-05-01:
I'm working locally with 0.0.17, I will push changes remote just in case of any relevant improvement. Sadly nothing so far.
I'm also testing the WDT approach specifically for ESP8266, let's see

matteocrippa, 2024-05-01:
I managed to capture the crash

```
22:47:11.527 > [I] Job: Random value: 1262036342
22:47:11.533 > [I] Job: Hex value: 4B392576
22:47:11.533 > [D] Job: Coinbase hash: 9906fbc5f64f7d7057ab41116b1b6289791a4f4ec1df1659ef8e530d5ea3a37b
22:47:11.543 > [D] Job: Merkle root: 9906fbc5f64f7d7057ab41116b1b6289791a4f4ec1df1659ef8e530d5ea3a37b
22:47:11.549 > [I] Current: Job: a0ed2c ready to be mined
22:47:11.554 > 
22:47:11.554 > User exception (panic/abort/assert)
22:47:11.560 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:47:11.565 > 
22:47:11.565 > Panic core_esp8266_main.cpp:191 __yield
22:47:11.565 > 
22:47:11.565 > >>>stack>>>
22:47:11.565 > 
22:47:11.565 > ctx: cont
22:47:11.571 > sp: 3ffff560 end: 3fffffc0 offset: 0010
22:47:11.571 > 3ffff570:  00000000 5d2f1a9f 3fff1310 40229e79  
22:47:11.634 > 3ffff580:  000007ff 3ffff5e8 3fff1310 40226b59  
22:47:11.724 > 3ffff590:  0000123c 00000020 00000000 40226c5d  
22:47:11.724 > 3ffff5a0:  3fff1310 00000020 3fffaf84 3fffb4e4  
22:47:11.724 > 3ffff5b0:  3fffb4fc 00000020 3fffaf84 3fffb4e4  
22:47:11.724 > 3ffff5c0:  3fffb4fc 3fffb514 00000205 40215ac5  
22:47:11.724 > 3ffff5d0:  3fffaf84 00000000 00000205 00000001  
22:47:11.724 > 3ffff5e0:  00000001 00004208 6469227b 756e3a22  
22:47:11.724 > 3ffff5f0:  222c6c6c 6874656d 3a22646f 6e696d22  
22:47:11.724 > 3ffff600:  2e676e69 69746f6e 2c227966 72617022  
22:47:11.724 > 3ffff610:  22736d61 61225b3a 32646530 222c2263  
22:47:11.724 > 3ffff620:  64366634 66346330 61326233 33663432  
22:47:11.724 > 3ffff630:  35656666 38643164 64343661 36663538  
22:47:11.724 > 3ffff640:  30393633 62383938 31303030 66323136  
22:47:11.724 > 3ffff650:  30303030 30303030 30303030 30303030  
22:47:11.724 > 3ffff660:  30222c22 30303032 30303030 30303031  
22:47:11.724 > 3ffff670:  30303030 30303030 30303030 30303030  
22:47:11.724 > 3ffff680:  30303030 30303030 30303030 30303030  
22:47:11.724 > 3ffff690:  30303030 30303030 30303030 30303030  
22:47:11.724 > 3ffff6a0:  30303030 30303030 30303030 66666630  
22:47:11.724 > 3ffff6b0:  66666666 30353166 64646433 35633037  
22:47:11.724 > 3ffff6c0:  34623436 35393432 34653234 34663433  
22:47:11.724 > 3ffff6d0:  222c2264 66666666 66666666 34383330  
22:47:11.724 > 3ffff6e0:  37313764 30303030 30303030 36313030  
22:47:11.724 > 3ffff6f0:  34313030 30653732 31656634 39613630  
22:47:11.724 > 3ffff700:  30396234 66306532 65633533 30653738  
22:47:11.724 > 3ffff710:  62313335 34666630 33336637 37386339  
22:47:11.724 > 3ffff720:  32313838 30303030 30303030 30303631  
22:47:11.724 > 3ffff730:  30393431 38316436 66363863 38336365  
22:47:11.724 > 3ffff740:  30316666 62303632 65396464 36653964  
22:47:11.724 > 3ffff750:  31653263 64663231 30303461 30303030  
22:47:11.724 > 3ffff760:  30303030 30303030 36323030 34326136  
22:47:11.724 > 3ffff770:  31326161 64653961 36663265 66336331  
22:47:11.724 > 3ffff780:  31643137 64666564 39616633 66643939  
22:47:11.724 > 3ffff790:  39363361 35373335 39366335 39383630  
22:47:11.728 > 3ffff7a0:  39393937 34623236 62656238 36333864  
22:47:11.728 > 3ffff7b0:  65343739 39666338 30303030 30303030  
22:47:11.733 > 3ffff7c0:  5d5b2c22 3032222c 30303030 2c223030  
22:47:11.739 > 3ffff7d0:  30373122 64313333 222c2262 32333636  
22:47:11.744 > 3ffff7e0:  65346161 61662c22 5d65736c 6632007d  
22:47:11.750 > 3ffff7f0:  30303030 30303030 30303030 30303030  
22:47:11.750 > 3ffff800:  30222c22 30303032 30303030 30303031  
22:47:11.755 > 3ffff810:  30303030 30303030 30303030 30303030  
22:47:11.761 > 3ffff820:  30303030 30303030 30303030 30303030  
22:47:11.766 > 3ffff830:  30303030 30303030 30303030 30303030  
22:47:11.771 > 3ffff840:  30303030 30303030 30303030 66666630  
22:47:11.771 > 3ffff850:  66666666 30353166 64646433 35633037  
22:47:11.777 > 3ffff860:  34623436 35393432 34653234 34663433  
22:47:11.782 > 3ffff870:  222c2264 66666666 66666666 34383330  
22:47:11.788 > 3ffff880:  37313764 30303030 30303030 36313030  
22:47:11.793 > 3ffff890:  34313030 30653732 31656634 39613630  
22:47:11.793 > 3ffff8a0:  30396234 66306532 65633533 30653738  
[... 182 lines cut ...]
22:47:14.176 > 0x40100e9a in free at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/umm_malloc/umm_malloc.cpp:688
22:47:14.176 > 0x402163fc in l_info(char const*, char const*, ...) at src/utils/log.cpp:30
22:47:14.176 > 0x40207f5a in miner(unsigned int) at src/miner/miner.cpp:47
22:47:14.176 > 0x4022a690 in esp_delay<__delay(long unsigned int)::<lambda()> > at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/coredecls.h:69
22:47:14.176 >  (inlined by) esp_delay<__delay(long unsigned int)::<lambda()> > at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/coredecls.h:78
22:47:14.176 >  (inlined by) __delay at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_wiring.cpp:39
22:47:14.176 > 0x402075be in loop at src/main.cpp:107
22:47:14.176 > 0x40229f76 in loop_wrapper() at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/core_esp8266_main.cpp:258
22:47:14.176 > 0x4010129d in cont_wrapper at /Users/matteocrippa/.platformio/packages/framework-arduinoespressif8266/cores/esp8266/cont.S:81
22:47:14.176 > 
22:47:14.176 > 
22:47:14.176 > --------------- CUT HERE FOR EXCEPTION DECODER ---------------
22:47:14.176 > 
22:47:14.176 >  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
22:47:14.176 > 
22:47:14.176 > load 0x4010f000, len 3424, room 16 
22:47:14.176 > tail 0
22:47:14.176 > chksum 0x2e
22:47:14.176 > load 0x3fff20b8, len 40, room 8 
22:47:14.178 > tail 0
22:47:14.178 > chksum 0x2b
22:47:14.178 > csum 0x2b
22:47:14.178 > v0009fcc0
22:47:14.178 > ~ld
```

ffrediani, 2024-05-02:
Hi Matteo
In fact 0.0.17 hasn't shown anyting much different. Actually it seems it lasts less time than 0.0.16. Whenever you push it we'll see again.

Regards the crash you shared yes I got similar ones, with the cut part for the decoder, for these ones it reboots and keeps going.
The main problem seems to be when this doesn't happen and it simply stops with no errors or anything to give a clue about. Let's see if a watchdog may help then.

Mobilpadde, 2025-06-27:
Any updates to this?