matteocrippa/leafminer #25: ESP8266 Re-subscribe and Auth when connection drops

wmikrut, 2024-03-23:
Occasionally the conenction will drop and re-connect.
The issue is that the connrection never re-subscribed and Authorized to the pool after that reconnection.

This leads to shares being submitted and and receiveng the following response:

03:29:43.307 > [I] Miner: [0] > [2495007] > 0x06135197 - diff 0.000022916033
03:29:43.307 > [I] Network: >>> {"id":1871,"method":"mining.submit","params":["wmikrut.ex1","2495007","206351","65fe5606","06135197"]}
03:29:43.323 > 
03:29:43.338 > [I] Network: <<< [mining.submit] {**"error": [-2, "Connection is not subscribed"**, "Traceback: <class 'mining_libs.stratum_listener.SubmitException'>: Connection is not subscribed\nstratum\\services.pyo:81:call\ntwisted\\internet\\defer.pyo:139:maybeDeferred\nstratum\\services.pyo:78:_run\ntwisted\\internet\\defer.pyo:1237:unwindGenerator\n--- <exception caught here> ---\ntwisted\\internet\\defer.pyo:1099:_inlineCallbacks\nmining_libs\\stratum_listener.pyo:181:submit\n"], "id": 1871, "result": null}
03:29:43.370 > [I] Network: Share accepted
03:29:43.370 > [I] Current: Hash accepted: 1694

The mining.submit accepts the request in network.cpp @ 302
else if (strcmp(type, "mining.submit") == 0)
    {
        l_info(TAG_NETWORK, "Share accepted");
        current_increment_hash_accepted();
    }
    
What we need in that statement is a check of the error code, when not null and is a specifc error code (-2) or the text "Connection is not subscribed" to call the Subscribe and Auth functions and skip reporing the Share was accepted because it was actuallt discarded by the pool.

matteocrippa, 2024-04-05:
@wmikrut is this still relevant with the latest version?

wmikrut, 2024-04-08:
Let me check... I've been on vacation for the past two weeks and I'm just getting caught up on everything,

matteocrippa, 2024-04-19:
@wmikrut do you have any update for me here?