matteocrippa/leafminer #26: Option to disable autoupdate

PSLLSP, 2024-04-08:
LeafMiner v 0.0.13

I would like to have an option to disable firmware autoupdate. Default option of checkbox could be **enable**

It looks like firmware can do  autoupdate:

```
$ miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I] Main: LeafMiner - v.0.0.13 - (C: 2)
[I] Main: ESP32 - Disable WDT
[I] Storage: Setup: OK
[I] AutoUpdate: Connecting to LuckyAP...
[D] AutoUpdate: payload: {
  "current": "0.0.13",
  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",
  "devices": [
    "esp8266",
    "esp32",
    "esp32-s2",
    "esp32-s3",
    "geekmagic-smalltv",
    "lilygo-t-display-s3"
  ]
}

[D] AutoUpdate: Remote Version: 0.0.13
[D] AutoUpdate: No Updates, Version: 0.0.13
[I] Network: Connecting to LuckyAP...
[D] Network: Connecting to host pool.vkbit.com...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.13", null]}
```

matteocrippa, 2024-04-11:
Feature added in the feature/v.0.0.14 branch

matteocrippa, 2024-04-22:
Added back in 0.0.15, I forgot to force html 2 header processing