matteocrippa/leafminer #27: Could be more details written to the console?

PSLLSP, 2024-04-08:
LeafMiner 0.0.13

I would like to see more details printed to serial console. I miss information like MAC address of the device, IP address assigned from DHCP server, device model, chipID, size of memory, stored configuration could be printed, etc.

Current output miss important details:

```
$ miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
[I] Main: LeafMiner - v.0.0.13 - (C: 2)
[I] Main: ESP32 - Disable WDT
[I] Storage: Setup: OK
[I] AutoUpdate: Connecting to LuckyAP...
[D] AutoUpdate: payload: {
  "current": "0.0.13",
  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",
  "devices": [
    "esp8266",
    "esp32",
    "esp32-s2",
    "esp32-s3",
    "geekmagic-smalltv",
    "lilygo-t-display-s3"
  ]
}

[D] AutoUpdate: Remote Version: 0.0.13
[D] AutoUpdate: No Updates, Version: 0.0.13
[I] Network: Connecting to LuckyAP...
[D] Network: Connecting to host pool.vkbit.com...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.13", null]}
...
```

I would like to see more information about connection to the AP and to the pool:

```
...
[I] Network: Connecting to LuckyAP...
[D] Network: Connecting to host pool.vkbit.com...
...
```
NerdMiner_V2 prints configuration (JSON file) during start and IP is reported too...

matteocrippa, 2024-04-11:
Added in `v/0.0.14`:

Info logs for
- configuration loaded
- wifi ip and mac address
- memory 
- chipID

matteocrippa, 2024-04-20:
v0.0.14 released, let me know @PSLLSP