matteocrippa/leafminer #29: ESP32-S3 with 4MB of FLASH cannot run LeafMiner

PSLLSP, 2024-04-09:
LeafMiner 0.0.13

[Mini module with ESP32-S3](https://www.aliexpress.com/item/1005006332489327.html), it has USB-C connector, 18 pins, on-board antenna, RGB LED and two buttons.

Device info:

```
$ esptool.py flash_id
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: dc:54:75:01:23:45
Uploading stub...
Running stub...
Stub running...
Manufacturer: 20
Device: 4016
Detected flash size: 4MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...
```

LeafMiner flashed with these commands:

```
esptool.py --baud 115200 write_flash 0x0000 bootloader_esp32-s3.bin
esptool.py --baud 115200  write_flash 0x8000 partitions_esp32-s3.bin
esptool.py --baud 115200  write_flash 0x10000 firmware_esp32-s3.bin
```

Device doesn't mine and I see in console that flash is corrupted:

```
$ miniterm /dev/ttyACM0 115200
...
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0x2b (SPI_FAST_FLASH_BOOT)
Saved PC:0x4037da2c
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (203) spi_flash: Detected size(4096k) smaller than the size in the binary image header(8192k). Probe failed.

assert failed: do_core_init startup.c:328 (flash_ret == ESP_OK)

Backtrace: 0x4037de1e:0x3fceb180 0x40383201:0x3fceb1a0 0x40389221:0x3fceb1c0 0x42021e62:0x3fceb2f0 0x4037d78f:0x3fceb320 0x403cd86b:0x3fceb350 0x403cdb2a:0x3fceb380 0x403c9925:0x3fceb4b0 0x40045c01:0x3fceb570 0x40043ab6:0x3fceb6f0 0x40034c45:0x3fceb710

ELF file SHA256: 9ed6ec27570d25a3

E (234) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
...
```

It is strange that 4MB of FLASH is not enough to run LeafMiner. NerdMiner_V2 1.6.3 was running at this device for some time...

matteocrippa, 2024-04-09:
I think there's something wrong in the flash procedure, but I was lately flashing using Platformio tool and it worked fine for me ( I have two ESP32-S3 running since a few weeks) worth double check

PSLLSP, 2024-04-11:
I found different module, it has 16 MB of FLASH. Module has ESP32-S3-WROOM-1 chip and two USB-C ports. I flashed it with the same procedure. It cannot mine but it complains with other error:

```
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
E (207) psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
```

The error message is in `firmware_esp32-s3.bin` file. The same applies to error `Core dump flash config is corrupted`:

```
$ strings firmware_esp32-s3.bin | grep "PSRAM chip not found"
E (%u) %s: PSRAM ID read error: 0x%08x, PSRAM chip not found or not supported, or wrong PSRAM line mode

$ strings firmware_esp32-s3.bin | grep "Core dump flash config is corrupted"
E (%u) %s: Core dump flash config is corrupted! CRC=0x%x instead of 0x%x
```

PSLLSP, 2024-04-11:
I finally find device that can be flashed with LeafMiner for ESP32-S3. LILYGO T-QT. It has 8MB of FLASH and it can run LeafMiner. Display is not used, it is blank. NerdMiner has version for this device, it uses display to show clock and hashrate, just very basic. I think this proof that the binary firmware in release cannot be run on device with only 4MB of FLASH...

This is LILYGO T-QT mini display:
  
```
$ esptool.py flash_id
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting...
Detecting chip type... ESP32-S3
Chip is ESP32-S3 (revision v0.1)
Features: WiFi, BLE
Crystal is 40MHz
MAC: f4:12:fa:01:22:10
Uploading stub...
Running stub...
Stub running...
Manufacturer: c8
Device: 4017
Detected flash size: 8MB
Flash type set in eFuse: quad (4 data lines)
Hard resetting via RTS pin...
```

matteocrippa, 2024-04-11:
But question, still relevant, did you try to build from source using platformio and use it to upload the firmware?
I think it can be that the issue is with with flash address at this point

PSLLSP, 2024-04-14:
I have not build LeafMiner from source code. I just tried to use binary files downloaded from "Release page". I assumed that I can use these binary files on any board that has compatible chip (ESP32-S3) but that is not the case. And that points to other issue. Documentation of LeafMiner doesn't specify what devices are compatible with release files; what was the development board used to test it, what is reference hardware...