matteocrippa/leafminer #30: ESP32 with 4MB of FLASH cannot run LeafMiner

PSLLSP, 2024-04-09:
LeafMiner v 0.0.13

Module with ESP32-WROOM-32 has , USB-C with HL-340 UART, 4MB of FLASH, 30 pins, 2 buttons, 2 LEDs. It can run NerdMiner_V2 (ESP32-devkit v 1.6.3) but LeafMiner doesn't run on this board.

```
$ esptool.py flash_id
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyUSB0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: 08:d1:f9:01:23:45
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

Flashed with LeafMiner 0.0.13:

```
# esptool.py  --baud 115200 write_flash -e 0 /dev/null # full chip ERASE
esptool.py --baud 115200 write_flash 0x0000 bootloader_esp32.bin
esptool.py --baud 115200 write_flash 0x8000 partitions_esp32.bin
esptool.py --baud 115200 write_flash 0x10000 firmware_esp32.bin
```

LeafMiner doesn't run:

```$ miniterm /dev/ttyUSB0 115200
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
invalid header: 0x04202040
ets Jul 29 2019 12:21:46

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
...
```

matteocrippa, 2024-04-09:
Same here, I think there's something wrong in the cli flasher.
I was testing using platformio on EP32, ESP32-S2 and ESP32-S3 and I never hit this issue, alternative can be an issue with the CI, but I don't think so

PSLLSP, 2024-04-10:
I modified offset of bootloader from 0x0000 to 0x1000 and I see some progress. I was able to connect to webUI at 192.168.4.1 and configure the miner but it cannot connect to my WiFi AP. NerdMiner was running on this module few minutes ago and it was working fine so I believe that WiFi configuration is correct and HW is not defective...

```
# esptool.py  --baud 115200 write_flash -e 0 /dev/null # full chip ERASE
esptool.py --baud 115200 write_flash 0x1000 bootloader_esp32.bin
esptool.py --baud 115200 write_flash 0x8000 partitions_esp32.bin
esptool.py --baud 115200 write_flash 0x10000 firmware_esp32.bin
```

This is what I see in the console, it looks like client cannot connect to WiFi... 
```
$ miniterm /dev/ttyACM0 115200
--- Miniterm on /dev/ttyACM0  115200,8,N,1 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jul 29 2019 12:21:46

configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13260
load:0x40080400,len:3028
entry 0x400805e4
E (265) psram: PSRAM ID read error: 0xffffffff
[I] Main: LeafMiner - v.0.0.13 - (C: 2)
[I] Main: ESP32 - Disable WDT
[I] Storage: Setup: OK
[I] Blink: Setup
[I] Blink: Blinking 5 times
[I] AutoUpdate: Connecting to LuckyAP...
```

This is different ESP32 module than last time, so I add output from `flash_id`

```
$ esptool.py flash_id
esptool.py v4.6.2
Found 1 serial ports
Serial port /dev/ttyACM0
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
Chip is ESP32-D0WD-V3 (revision v3.1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: 08:d1:f9:01:23:56
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
```

matteocrippa, 2024-04-19:
Btw checkin the verbose log of Platfomio, it flashes it in this way:

```
esptool.py --chip esp32 --port "/dev/cu.Bluetooth-Incoming-Port" --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x1000 bootloader.bin 0x8000 partitions.bin 0xe000 .platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin 0x10000 .pio/build/esp32/firmware.bin
```

PSLLSP, 2024-04-22:
It is interesting that PlatformIO flashes some file to address 0xe000 (`0xe000 .platformio/packages/framework-arduinoespressif32/tools/partitions/boot_app0.bin`) but that file `boot_app0.bin` is not included in release files... NerdMiner has 4 files in release folder and 0xe000_boot_app0.bin is in the list...

matteocrippa, 2024-04-22:
I added that file to the CI script, let's see if helps.
An extra plan would be to prepare a web flasher, but I have not that time right now

PSLLSP, 2024-04-23:
I see that version 0.0.15 was released and I see new files in release.
I am curious about files `0xe000_boot_esp8266.bin` and `0xe000_boot_geekmagic-smalltv.bin`. I assume those should not be there. Please, verify in log file from PlatformIO...

Webflasher is great help for new users but I do not think it is priority at this point. LeafMiner is not ready for "mass production". I used NerdMiner and it works but there was no development for several months and even simple issue with pool password was not fixed so I was looking for an alternative. I have found LeafMiner. I still see it is not ready, so I have not replaced any NerdMiner with LeafMiner and I cannot recommend anyone to switch. That is why I think that webflasher is not priority. Even webflasher for NerdMiner is not without issues, some boards just cannot be flashed with their webflasher. `esptool` can flash all my boards...

MajorDomDePIN, 2025-01-17:
Is there any new info about the web flasher for esp8266? 

i tried it multiple times to flash the LeafMiner/NerdMiner onto my esp8266 but sadly it never worked properly. 

[image]

i tried to flash it multiple times with '0000' and '1000' as offset but this didn't helped either

When I once flashed with NerdMiner (wrong firmware/sp32), at least the LED stayed on all the time after rebooting after flashing, but the AccessPoint could still not be found. 

Unfortunately I only failed when trying to flash the LeafMiner firmware

matteocrippa, 2025-01-17:
Did you try just to flash this: [0x0000_firmware_esp8266.bin](https://github.com/matteocrippa/leafminer/releases/download/v0.0.16/0x0000_firmware_esp8266.bin)
At 0x0000?

MajorDomDePIN, 2025-01-17:
hey, thanks for the fast response!

i just tried it this way and now the AP shows up, so are the other .bin-files not necessary?

[image]

matteocrippa, 2025-01-17:
for esp8266 no, just that one

MajorDomDePIN, 2025-01-17:
[image]

Wheeey :) after a few more attempts it's running fine! 

Ty <3

MajorDomDePIN, 2025-01-18:
i could solve it for the most, but i still have the issue that after a few minutes the worker isn't listed on public-pool anymore:

[image]

the first 11 hashes got accepted, the 12th was rejected, then accepted.. it accepts 3 more but then the session stops to get showen on public-pool.io, and the wlan connection starts to reset all the time. 

only when i fully restart the esp8266 the worker starts to show up again on public-pool and mines fine for around 5 minutes until it breaks down again

MajorDomDePIN, 2025-01-22:
sadly i still have the issue that after a certain time the esp goes into a loop where it reconnects to the wlan and also shows on the terminal that it's trying to mine but it wont get showed on public pool anymore, just when i reboot the esp its there again for some time (minutes to hours) until its stuck again. 

Can provide a screenshot of the log later if wanted