matteocrippa/leafminer #31: Add address to firmware files

PSLLSP, 2024-04-09:
It would be nice to have address of FLASH memory in the the firmware filename. It helps beginners, it is self documenting feature.

EXAMPLE.

Current filenames of firmware for LeafMiner 0.0.13 for ESP32:

```
bootloader_esp32.bin
partitions_esp32.bin
firmware_esp32.bin
```

Better filenames:

```
0x0000_bootloader_esp32.bin
0x8000_partitions_esp32.bin
0x10000_firmware_esp32.bin
```

Other point is that flashing of firmware could be scripted and when offset address is stored in filename, it is easier to design script...

matteocrippa, 2024-04-11:
I will need to extract them from the platformio logs, but have not time for that in short period, any help or PR will be appreciated

tiger0007t, 2024-04-19:
> Было бы неплохо в имени файла прошивки указать адрес FLASH-памяти. Это помогает новичкам, это функция самодокументирования.
> 
> ПРИМЕР.
> 
> Текущие имена файлов прошивки LeafMiner 0.0.13 для ESP32:
> 
> ```
> bootloader_esp32.bin
> partitions_esp32.bin
> firmware_esp32.bin
> ```
> 
> Лучшие имена файлов:
> 
> ```
> 0x1000_bootloader_esp32.bin
> 0x8000_partitions_esp32.bin
> 0x10000_firmware_esp32.bin
> ```
> 
> Другой момент заключается в том, что прошивку прошивки можно выполнить по сценарию, и когда адрес смещения хранится в имени файла, создать сценарий проще...
> 
> Кстати, я только что угадал адреса в своем примере, возможно, я использовал адреса неправильные...
> 
> Это список предлагаемых имен файлов, работа в процессе. ESP32 и ESP8266 были проверены на корректность. У меня нет устройства ESP-S2 и GeekMagic-smallTV (я предполагаю, что оно основано на ESP8266).
> 
> ```
> 0x0000_firmware_esp8266.bin
> 
> 0x0000_firmware_geekmagic-smalltv.bin
> 
> 0x1000_bootloader_esp32.bin
> 0x8000_partitions_esp32.bin
> 0x10000_firmware_esp32.bin
> 
> 0x0000_bootloader_esp32-s2.bin
> 0x8000_partitions_esp32-s2.bin
> 0x10000_firmware_esp32-s2.bin
> 
> 0x0000_bootloader_esp32-s3.bin
> 0x8000_partitions_esp32-s3.bin
> 0x10000_firmware_esp32-s3.bin
> 
> 0x0000_bootloader_lilygo-t-display-s3.bin
> 0x8000_partitions_lilygo-t-display-s3.bin
> 0x10000_firmware_lilygo-t-display-s3.bin
> ```

> Было бы неплохо в имени файла прошивки указать адрес FLASH-памяти. Это помогает новичкам, это функция самодокументирования.
> 
> ПРИМЕР.
> 
> Текущие имена файлов прошивки LeafMiner 0.0.13 для ESP32:
> 
> ```
> bootloader_esp32.bin
> partitions_esp32.bin
> firmware_esp32.bin
> ```
> 
> Лучшие имена файлов:
> 
> ```
> 0x1000_bootloader_esp32.bin
> 0x8000_partitions_esp32.bin
> 0x10000_firmware_esp32.bin
> ```
> 
> Другой момент заключается в том, что прошивку прошивки можно настроить по сценарию, и когда адрес смещения сохраняется в имени файла, создать скрипт проще...
> 
> Кстати, я только что угадал адрес в своем почтовом ящике, возможно, я использовал неправильные адреса...
> 
> Это список предлагаемых названий файлов, работа в процессе. ESP32 и ESP8266 были проверены на корректность. У меня нет устройства ESP-S2 и GeekMagic-smallTV (я предполагаю, что оно основано на ESP8266).
> 
> ```
> 0x0000_firmware_esp8266.bin
> 
> 0x0000_firmware_geekmagic-smalltv.bin
> 
> 0x1000_bootloader_esp32.bin
> 0x8000_partitions_esp32.bin
> 0x10000_firmware_esp32.bin
> 
> 0x0000_bootloader_esp32-s2.bin
> 0x8000_partitions_esp32-s2.bin
> 0x10000_firmware_esp32-s2.bin
> 
> 0x0000_bootloader_esp32-s3.bin
> 0x8000_partitions_esp32-s3.bin
> 0x10000_firmware_esp32-s3.bin
> 
> 0x0000_bootloader_lilygo-t-display-s3.bin
> 0x8000_partitions_lilygo-t-display-s3.bin
> 0x10000_firmware_lilygo-t-display-s3.bin
> ```

Good afternoon, I need your help, I have several esp8266(esp12e) devices and I would really like to put them into mining. please explain how to do this by programming devices through an online program, for the esp32 series there are bootloader files and others on the site, and for the esp8266 there is only a firmware version. how to flash correctly at addresses 0x0000, 0x8000, 0x10000, I really ask for help, because programming only the firmware program esp12e goes into error after a while, judging by the UART port. And there is another question for the developer, I can't find how to contact him, as far as possible to make sure that there is one esp01 device, for example, and processors such as stm32f103 are connected to it (this is an example) how is it possible to create a kind of hash board with a lot of stm? in order not to overload the Internet router too much, is the idea itself possible? I'm a ge programmer to be able to redo the code for me, it's still difficult, thanks for the answer in advance. The STM32 is cheaper than the ESP series.

tiger0007t, 2024-04-19:
[image]
Where can I get the esp8266 programming files at addresses 0x0000, 0x8000, they are not in the repository, and what to register at address 0xE000.

matteocrippa, 2024-04-19:
@tiger0007t for esp8266 you don't need them

matteocrippa, 2024-04-20:
@PSLLSP https://github.com/matteocrippa/leafminer/releases/tag/v0.0.14

PSLLSP, 2024-04-21:
Autoupdate of v0.0.13 to 0.0.14 failed. I assume it is related to change of filename format...

This is ESP8266, v0.0.13:

```
...
[D] AutoUpdate: Remote Version: 0.0.14
[D] AutoUpdate: New Version: 0.0.14
[D] AutoUpdate: Device supported: esp8266
[D] AutoUpdate: Downloading: https://github.com/matteocrippa/leafminer/releases/download/v0.0.14/firmware_esp8266.bin
[E] AutoUpdate: httpCode: 404
```

I tried to flash firmware 0.0.14 directly but it doesn't work, I see exception:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0009f5b0
~ld
���␀�8␂�n��>␒r���n�b␌␘�␌␜�␌␜r␘␂␜b�␛␒�␀␌��␒�␇��␀l�<␒�␄�n�␃��l`␛�>␜r�8␛��l␒ #␌␘�␌�[I] Main: LeafMiner - v.0.0.14 - (C: 1)
[I] Main: Compiled: Apr 20 2024 15:11:44
[I] Main: Free memory: 39640
[I] Main: 
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x40237901 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00777220 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffc40 end: 3fffffc0 offset: 0150
3ffffd90:  00777221 3ffe8368 3ffffdf0 402358fd  
3ffffda0:  00000005 feefeffe feefeffe 3ffe8368  
3ffffdb0:  00000009 3ffe8368 4023a280 4023a344  
3ffffdc0:  00000000 3ffe8a8c 00000000 00000000  
3ffffdd0:  feefeffe feefeffe feefeffe 3ffe8a96  
3ffffde0:  3ffe8a97 3ffe8368 3ffffea0 4023a794  
3ffffdf0:  00000000 ffffffff 00000000 00000000  
3ffffe00:  00000005 00000009 fe302073 feefeffe  
3ffffe10:  feefeffe feefeffe feefeffe feefeffe  
3ffffe20:  feefeffe feefeffe feefeffe 3933effe  
3ffffe30:  00303436 feefeffe feefeffe feefeffe  
3ffffe40:  feefeffe feefeffe 00000000 feefeffe  
3ffffe50:  3fffff80 3fffff60 0000000c feefeffe  
3ffffe60:  3fffff80 3fffff60 00000008 00000005  
3ffffe70:  00000000 feefeffe feefeffe feefeffe  
3ffffe80:  feefeffe feefeffe feefeffe 3ffe8a8c  
3ffffe90:  3fffdad0 3ffe8368 00000000 402378b9  
3ffffea0:  00000000 feefeffe 00000000 ffff0208  
3ffffeb0:  00000000 00000000 feefeffe feefeffe  
3ffffec0:  feefeffe feefeffe feefeffe feefeffe  
3ffffed0:  feefeffe feefeffe feefeffe 4022b521  
3ffffee0:  00000002 eb926f86 feefef00 4022b521  
3ffffef0:  00000012 feefeffe feefeffe 4022b521  
3fffff00:  00000002 3fff1210 3ffeac3c 402253dc  
3fffff10:  0000012d 3ffe85d8 3fff1210 402378f5  
3fffff20:  3fffff80 3fffff60 00000008 402253dc  
3fffff30:  402253d0 3fff1210 3ffea568 402265ed  
3fffff40:  3fffdad0 3ffe85d8 3fff1210 402160a4  
3fffff50:  3fffff80 3fffff60 00000008 402254d9  
3fffff60:  000005dc feefeffe 00777221 00000021  
3fffff70:  00000000 00000000 0000004e 3fff13f4  
3fffff80:  3fffdad0 0000004e 000005dc 3fff13f4  
3fffff90:  3fffdad0 00000000 3ffe85d8 40207705  
3fffffa0:  feefeffe feefeffe 3fffdab0 40229c27  
3fffffb0:  feefeffe feefeffe feefeffe 4010129d  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
```

matteocrippa, 2024-04-21:
Autoupdate should be fixed now, and the crash seems related to the chipID, but I have no esp8266 board next to me to test it, let me just comment it out and force building again 0.0.14

PSLLSP, 2024-04-22:
I checked the device and it was already updated to 0.0.14 and mining. I confirm that autoupdate issue was fixed and the exception was fixed too.

tiger0007t, 2024-04-22:
> Я проверил устройство и оно уже обновилось до 0.0.14 и майнинг. Я подтверждаю, что проблема с автообновлением устранена, а также исправлено исключение.

tell me, does it work stably for you? it runs for 8-10 minutes and stops completely.

matteocrippa, 2024-04-22:
I was testing both 0.0.14 and 0.0.15 with Weimos D1 right now and it worked fine, I was still with serial console open

tiger0007t, 2024-04-22:
02:44:08.562 > [I] Network: <<< [mining.submit] {"id":68,"error":null,"result":true}
02:44:08.562 > [I] Network: Share accepted
02:44:08.577 > [I] Current: Hash accepted: 52
02:44:13.560 > [D] Current: Hashrate: 0.14 kH/s
02:44:15.291 > [I] Miner: [0] > [1d9aa01] > 0x006bfc1d - diff 0.000018099848
02:44:15.291 > [I] Network: >>> {"id":69,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1d9aa01","15609527","6626f3f0","006bfc1d"]}
02:44:15.309 >
02:44:15.623 > [I] Network: <<< [mining.submit] {"id":69,"error":null,"result":true}
02:44:15.623 > [I] Network: Share accepted
02:44:15.639 > [I] Current: Hash accepted: 53
02:44:20.618 > [D] Current: Hashrate: 5.12 kH/s
02:44:22.046 > [I] Miner: [0] > [1d9aa01] > 0x006c7103 - diff 0.000020738957
02:44:22.062 > [I] Network: >>> {"id":70,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1d9aa01","15609527","6626f3f0","006c7103"]}
-----------------------------------
at first everything goes well, then such a mistake pops up
-----------------------------------
02:44:28.817 > [I] Network: <<< [unknown] {"id":null,"method":"mining.notify","params":["1d9e59f","9278a2c18a7da5a866de736bfe6102c484f4069300023e1a0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff1703dfd20c5075626c69632d506f6f6c","ffffffff02346b501b0000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed2daa1cebf11699749368ea284301c91cf5d062409feb974ad4fdf3ad2d5706d100000000",["de6ec7b1cb5fdd52eda206a74ad1c713e774839246312d74181c87d1305ad90b","09713b0e68da329092fb35329801bec20304419996cb7bf3cc
02:44:28.866 > [E] Network: Unknown response type: unknown
-----------------------------------
and after it fixates on it
-----------------------------------
02:47:42.408 >
02:47:47.388 > [D] Current: Hashrate: 3.02 kH/s
02:47:49.973 > [I] Miner: [0] > [1d9aa01] > 0x00865bce - diff 0.001734373647
02:47:49.991 > [I] Network: >>> {"id":94,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1d9aa01","15609527","6626f3f0","00865bce"]}
02:47:50.007 >
02:47:51.954 > [I] Network: <<< [unknown] 37a77fcbdfcc1c","6ad1a1960c9a29d6393c0e1e7bbf6eaa011fb8b6a98bf6b557cc8a8d73c26007","7c98a6741262fe20d2323b56ddd95b8572fabcf2d6f49e31f2054b28eb4fe986","695bf6394a7b813551dba08cd6b24da1815048e806b5081415f8d7672052d8c0","15f8ac9b0c49a7885dcf9dd268422ef938e22530bae8ee05a9e6b4333c4c494d","2c00002a2b8bb08abf9b686e45559762c589c9332cb2656be6e4305dc95314df","ce8236e7cec1216846d1a76093a3c95685ac2ff59ad0dd5bfd58cbeaf594ea84","ec34e11a0c2c89d914f98a3003411c1a59acf906317d9ad365a881c50ff0a183","5074c9c6bfb0b7a06eac3fbfbabd1def2ffd18ab62e9f4c5a2e278170843e980","0a4af037080b12dd246bfac7921d158451322bea90e84a9af4f6a5a701995bd4","0baebd4ab8c08b255613d0ff300d939d86b65ebc6f37d36ac1c17cfaa4718d8a","44085990d0cc286da7ebd42ae10e8befb5540ea16848613ec5a9b2990bbe4506"],"20000000","17034219","6626f695",true]}
02:47:52.022 > [E] Network: Unknown response type: unknown
02:47:52.032 > [I] Network: <<< [mining.submit.fail] {"id":70,"result":null,"error":[21,"Job not found",""]}
02:47:52.032 > [E] Network: Share rejected
02:47:52.032 > [I] Current: Hash rejected: 1
02:47:52.032 > [I] Network: <<< [mining.submit.fail] {"id":71,"result":null,"error":[21,"Job not found",""]}
02:47:52.048 > [E] Network: Share rejected
02:47:52.048 > [I] Current: Hash rejected: 2
02:47:52.048 > [I] Network: <<< [mining.submit.fail] {"id":72,"result":null,"error":[21,"Job not found",""]}
02:47:52.063 > [E] Network: Share rejected
02:47:52.063 > [I] Current: Hash rejected: 3
------------------------------
then tries to reconnect and find the job

PSLLSP, 2024-04-23:
ESP8266 with firmware 0.0.13 fails to autoupdate:

```
[D] AutoUpdate: Remote Version: 0.0.15
[D] AutoUpdate: New Version: 0.0.15
[D] AutoUpdate: Device supported: esp8266
[D] AutoUpdate: Downloading: https://github.com/matteocrippa/leafminer/releases/download/v0.0.15/firmware_esp8266.bin
[E] AutoUpdate: httpCode: 404
[I] Network: Connecting to LuckyAP...
[D] Network: Connecting to host pool.vkbit.com...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.13", null]}
```

matteocrippa, 2024-04-29:
@PSLLSP I think now it should be fine, both for naming and auto updating, isn't it?
If so can we close this?

PSLLSP, 2024-04-30:
It is not OK...

ESP8266 with 0.0.13 cannot be updated to 0.0.16:

```
[I] Main: ESP8266 - Disable WDT
[I] Storage: Setup: OK
[I] Blink: Setup
[I] Blink: Blinking 5 times
[I] AutoUpdate: Connecting to LuckyAP...
[D] AutoUpdate: payload: {
  "current": "0.0.16",
  "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",
  "devices": [
    "esp8266",
    "esp32",
    "esp32-s2",
    "esp32-s3",
    "geekmagic-smalltv",
    "lilygo-t-display-s3"
  ]
}

[D] AutoUpdate: Remote Version: 0.0.16
[D] AutoUpdate: New Version: 0.0.16
[D] AutoUpdate: Device supported: esp8266
[D] AutoUpdate: Downloading: https://github.com/matteocrippa/leafminer/releases/download/v0.0.16/firmware_esp8266.bin
[D] AutoUpdate: Downloaded: 653232
[D] AutoUpdate: Begin Update
[E] AutoUpdate: Written: 0
[I] Network: Connecting to LuckyAP...
[D] Network: Connecting to host pool.vkbit.com...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.13", null]}
```