matteocrippa/leafminer #32: Enabling Cascade-Mining

IdiocracyFR, 2024-04-14:
Hello, I have been looking at this for about a week. It's been a lot of fun, but I'm stuck scratching my head. 

I cant for the life of me figure out how to enable cascade-mining and build firmware for branches and leaves (or if I would need to).

I've been trying with both esp266 & esp32 modules to no avail. 
Using vscode w/ the platformio extension I have been able to configure the firmware & flash modules successfully, but nothing from serial monitor indicated that it was working in any particular mode, or communicating with the esps. 
Looking through the source code, I haven't been able to find what I need to add to the configuration in order to enable cascade-mining, or what else I might need to do.

If you could set me on the right path here, it would be much appreciated!

Cheers :)

matteocrippa, 2024-04-15:
Cascade-mining is not available yet, I had to steer my focus on solo mining fist

IdiocracyFR, 2024-04-15:
That explains it! lol 
Appreciate you getting back to me. Maybe I'll be able to work out something similar.

closed 2024-04-15 by IdiocracyFR

matteocrippa, 2024-04-15:
In general it's more a matter to introduce MQTT protocol and have just one device to connect to stratum. But not that much time to take care of that in near future... but any PR is always welcome

matteocrippa, 2024-04-15:
In this scenario, probably even a basic socket can be good.
But ESP-now can be an option too if we consider the BLE communication, you can even consider using a GATT server... worth trying the esp-now, just not sure it introduces slowdown in the hashing

IdiocracyFR, 2024-04-16:
A friend and I are going to mess with it. I'll make a fork if we come up
with anything interesting :)