matteocrippa/leafminer #33: Esp8266 & STM32 or other processors

tiger0007t, 2024-04-19:
Good afternoon, please tell me is it possible to make it so that there is one esp8266 device to which several other processors are connected to work on calculations? So that the esp8266 issues tasks for work, as it happens in hash boards on ASICs, for example, stm32 or atmega, or others. This is a very interesting topic. This would make it possible to assemble a device like an asic mini only. The cost of the stm32 and its power consumption are noticeably lower than those of the esp32 or esp8266

matteocrippa, 2024-04-19:
There's a plan for that, but I don't have any ETA for nor development or release.
It's described here: https://github.com/matteocrippa/leafminer/blob/main/docs/cascade-mining.md

A few issue with solo mining needs to be addressed first than we can expand to that feature.
For now no plan to use STM.

tiger0007t, 2024-04-19:
where can I download version 0.0.14 for testing?

matteocrippa, 2024-04-19:
0.0.14 has not been released yet, the fastest way is to install it using vscode+platformio

tiger0007t, 2024-04-19:
I have vscode, and platformio is installed, but it's difficult for me to work in it, I didn't have to do something in it, how to add a project doesn't even always work, you need to figure it out. You can suggest on the next branch I asked where to get the files for esp8266
https://github.com/matteocrippa/leafminer/issues/31#issuecomment-2066309571

tiger0007t, 2024-04-19:
[image]
likely it is necessary to refuse esp8266 doesn't leave to compile at me the project, something loads and is long strong, it was necessary to establish GIT control why he is necessary didn't understand as he participates in updates with github, on esp32 nerdminer_v2 works perfectly around the clock. there was a strong wish to start it on esp12e. knowledge it isn't enough to complete of it.

matteocrippa, 2024-04-19:
I didn't get, did you try to connect the board and then select Build & Upload?
I see you are in master branch

tiger0007t, 2024-04-20:
good afternoon, I'm trying to compile the project, but it takes a very long time to download packages from github, please tell me correctly I understood that it compiles for all boards?

tiger0007t, 2024-04-20:
Выполнение задачи: C:\Users\tiger0007\.platformio\penv\Scripts\platformio.exe run --target upload 

Processing esp8266 (platform: https://github.com/platformio/platform-espressif8266.git; board: d1_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (4.2.1+sha.87fdc0f) > WeMos D1 R2 and mini
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30102.0 (3.1.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)    
 - tool-mklittlefs @ 1.203.210628 (2.3)       
 - tool-mkspiffs @ 1.200.0 (2.0)
 - toolchain-xtensa @ 2.100300.220621 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- TridentTD_ESP_TrueRandom @ 1.0.0+sha.0ff7623
|-- cJSON @ 0.0.0+20240419220526.sha.87d8f09    
|-- ESP Async WebServer @ 1.2.3+sha.f71e3d4     
|-- Preferences @ 2.1.0
|-- DNSServer @ 1.1.1
|-- ESP8266mDNS @ 1.2
|-- ESP8266WiFi @ 1.0
|-- ESP8266HTTPClient @ 1.2
Building in debug mode
Retrieving maximum program size .pio\build\esp8266\firmware.elf
Checking size .pio\build\esp8266\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=====     ]  48.0% (used 39360 bytes from 81920 bytes)
Flash: [======    ]  61.8% (used 645956 bytes from 1044464 bytes)
Configuring upload protocol...
AVAILABLE: espota, esptool
CURRENT: upload_protocol = esptool
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\esp8266\firmware.bin
esptool.py v3.0
Serial port COM5
Connecting........
Chip is ESP8285
Features: WiFi, Embedded Flash
Crystal is 26MHz
MAC: ff:ff:ff:ff:ff:ff
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Compressed 650112 bytes to 413845...
Writing at 0x00000000... (3 %)
Writing at 0x00004000... (7 %)
Writing at 0x00008000... (11 %)
Writing at 0x0000c000... (15 %)
Writing at 0x00010000... (19 %)
Writing at 0x00014000... (23 %)
Writing at 0x00018000... (26 %)
Writing at 0x0001c000... (30 %)
Writing at 0x00020000... (34 %)
Writing at 0x00024000... (38 %)
Writing at 0x00028000... (42 %)
Writing at 0x0002c000... (46 %)
Writing at 0x00030000... (50 %)
Writing at 0x00034000... (53 %)
Writing at 0x00038000... (57 %)
Writing at 0x0003c000... (61 %)
Writing at 0x00040000... (65 %)
Writing at 0x00044000... (69 %)
Writing at 0x00048000... (73 %)
Writing at 0x0004c000... (76 %)
Writing at 0x00050000... (80 %)
Writing at 0x00054000... (84 %)
Writing at 0x00058000... (88 %)
Writing at 0x0005c000... (92 %)
Writing at 0x00060000... (96 %)
Writing at 0x00064000... (100 %)
Wrote 650112 bytes (413845 compressed) at 0x00000000 in 36.4 seconds (effective 142.9 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
========================================================== [SUCCESS] Took 54.58 seconds ==========================================================

if I understood everything correctly, then programming in vscode was successful.
-----------------------------------------------------------
Per processor
Processing esp32-s2 (platform: https://github.com/platformio/platform-espressif32.git#develop; board: lolin_s2_mini; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------Tool Manager: Installing espressif/toolchain-xtensa-esp32s2 @ 8.4.0+2021r2-patch5
Downloading  [------------------------------------]    0%

cannot load libraries yet
---------------------------------------------------------------

please tell me it's okay that vscode is trying to write the next firmware from another processor after programming?

tiger0007t, 2024-04-20:
12:29:03.043> [I] Current: Hash accepted: 11
12:29:10.072> [D] Current: Hashrate: 11.34 kH/s
12:29:15.962> [D] Current: Hashrate: 21.02 kH/s
12:29:17.087> [D] Current: Hashrate: 21.01 kH/s
12:29:18.782> [D] Current: Hashrate: 20.99 kH/s
12:29:21.208> [D] Current: Hashrate: 21.01 kH/s
12:29:26.287> [I] Miner: [0] > [1a8cdb2] > 0x004097ae - diff 0.000517413331
12:29:26.287> [I] Network: >>> {"id":20,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1a8cdb2","55880833","662389fd","004097ae"]}
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1a8e614","19a550d4bb16cfabed08dbfb49c8beb17c397301000237200000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170379d10c5075626c69632d506f6f6c","ffffffff02ba17aa490000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed85c8e58d7e3a6c1112f0146feb9e1c6a990853dd98bae9a2796b018059f51d6b00000000",["39852936aadffa1a0010e6d09e6b80cfec64108ab805cbed71df1e6bbfaecef7","fe9c5d7609a95cfac70306865bd8a25099cf5f5311c90638142ad250d8aae092","42a687025308c506feb09af2e5429a8dd9f7912:29:26.586> 0ed315ae52ca2019ca6b87fd904","13ca607481ba69a2047288fa160e56da081aebb2b0fd90f609183f1a45f467c4","d021975881d3ab193220d4da646129563b35918cd49237014aa16fc52e474b97","9a8529e8434fdf3f8e6379775c584ab8484aebd40861cec86c4bac959d8c3301","e4757a5ecc4095734351bfc0c17b872f6edefc9c708301283b6182c326f4f2c9","a70eab5bd55ba6520baee5058377420aa96d85a1af3740cfd0d6adc8e48ef071","4383748d83bea6821e04527b96d0a4f8b6095159c95ba2d81876fefccde4ca39","f4e5cf8b4f6dcb8e89be63d3c9ec09207a0a8a831a2734c69e76af4ec1cc01d8","a36b19481c1192a0c1afc8e19b9a08f43a2ab77d5a43cd35ceb3dab323a69718","33fa8c7adaa4daca72414d704e22c2896f602c395e01a0f70127f9cba4dd77b9","5fc9439c974dab82d39f4f751f5badfbdbca90912fdd4e7b3a0e4650662d0963"],"20000000","17034219","66238b29",false]}
12:29:26.586> [I] Job: Random value: 1784029717
12:29:26.586> [I] Current: Job: 1a8e614 queued
12:29:26.586> [I] Network: <<< [mining.submit] {"id":20,"error":null,"result":true}
12:29:26.586> [I] Network: Share accepted
12:29:26.586> [I] Current: Hash accepted: 12
12:29:33.566> [D] Current: Hashrate: 12.10 kH/s
12:29:35.753> [D] Current: Hashrate: 21.01 kH/s
12:29:39.139> [D] Current: Hashrate: 21.01 kH/s
12:29:41.449> [D] Current: Hashrate: 21.01 kH/s
12:29:48.375> [D] Current: Hashrate: 21.02 kH/s
12:29:57.693> [D] Current: Hashrate: 21.02 kH/s
12:29:58.891> [D] Current: Hashrate: 21.01 kH/s
12:30:01.263> [I] Miner: [0] > [1a8cdb2] > 0x004a20aa - diff 0.000374922019
12:30:01.263> [I] Network: >>> {"id":22,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1a8cdb2","55880833","662389fd","004a20aa"]}
12:30:01.263> [I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.000032]}
12:30:01.263> [I] Current: New difficulty: 0.000032000000
12:30:01.263> [D] Network: Difficulty set to: 0.0000320000
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1a8e636","19a550d4bb16cfabed08dbfb49c8beb17c397301000237200000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170379d10c5075626c69632d506f6f6c","ffffffff02ba17aa490000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed85c8e58d7e3a6c1112f0146feb9e1c6a990853dd98bae912:30:01.949> a2796b018059f51d6b00000000",["39852936aadffa1a0010e6d09e6b80cfec64108ab805cbed71df1e6bbfaecef7","fe9c5d7609a95cfac70306865bd8a25099cf5f5311c90638142ad250d8aae092","42a687025308c506feb09af2e5429a8dd9f790ed315ae52ca2019ca6b87fd904","13ca607481ba69a2047288fa160e56da081aebb2b0fd90f609183f1a45f467c4","d021975881d3ab193220d4da646129563b35918cd49237014aa16fc52e474b97","9a8529e8434fdf3f8e6379775c584ab8484aebd40861cec86c4bac959d8c3301","e4757a5ecc4095734351bfc0c17b872f6edefc9c708301283b6182c326f4f2c9","a70eab5bd55ba6520baee5058377420aa96d85a1af3740cfd0d6adc8e48ef071","4383748d83bea6821e04527b96d0a4f8b6095159c95ba2d81876fefccde4ca39","f4e5cf8b4f6dcb8e89be63d3c9ec09207a0a8a831a2734c69e76af4ec1cc01d8","a36b19481c1192a0c1afc8e19b9a08f43a2ab77d5a43cd35ceb3dab323a69718","33fa8c7adaa4daca72414d704e22c2896f602c395e01a0f70127f9cba4dd77b9","5fc9439c974dab82d39f4f751f5badfbdbca90912fdd4e7b3a0e4650662d0963"],"20000000","17034219","66238b29",false]}
12:30:01.949> [I] Job: Random value: 594468053
12:30:01.949> [I] Current: Job: 1a8e636 queued
12:30:02.233> [I] Network: <<< [mining.submit] {"id":22,"error":null,"result":true}
12:30:02.233> [I] Network: Share accepted
12:30:02.233> [I] Current: Hash accepted: 13
12:30:08.450> [D] Current: Hashrate: 8.29 kH/s
12:30:09.265> [I] Miner: [0] > [1a8cdb2] > 0x004ad6c4 - diff 0.000045173565
12:30:09.265> [I] Network: >>> {"id":24,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1a8cdb2","55880833","662389fd","004ad6c4"]}
12:30:09.516> [I] Network: <<< [mining.submit] {"id":24,"error":null,"result":true}
12:30:09.516> [I] Network: Share accepted
12:30:09.516> [I] Current: Hash accepted: 14
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1a8eb03","19a550d4bb16cfabed08dbfb49c8beb17c397301000237200000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170379d10c512:30:14.344> 075626c69632d506f6f6c","ffffffff02ba17aa490000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed85c8e58d7e3a6c1112f0146f [... cut ...] 6d0a4f8b6095159c95ba2d81876fefccde4ca39","f4e5cf8b4f6dcb8e89be63d3c9ec09207a0a8a831a2734c69e76af4ec1cc01d8","a36b19481c1192a0c1afc8e19b9a08f43a2ab77d5a43cd35ceb3dab323a69718","33fa8c7adaa4daca72414d704e22c2896f602c395e01a0f70127f9cba4dd77b9","5fc9439c974dab82d39f4f751f5badfbdbca90912fdd4e7b3a0e4650662d0963"],"20000000","17034219","66238b65",false]}
12:30:14.344> [I] Job: Random value: 456877666
12:30:14.344> [I] Current: Job: 1a8eb03 queued

stops and does not send anything to uart unfortunately

matteocrippa, 2024-04-20:
I can't see what version is running? 0.0.13 or 0.0.14?
Btw I see you managed to make it work now from source

tiger0007t, 2024-04-20:
I started compiling in vscode, I do not know which version he installed in the processor. there is a compilation log with library versions on top. after a short period of time, the esp8266 stops working

matteocrippa, 2024-04-20:
Can you just open the serial monitor and reboot the board, it should connect and show you which version is running

tiger0007t, 2024-04-21:
13:54:28.039 -> Ь⸮#[I] AutoUpdate: Connecting to DRE2306...
13:54:33.114 -> [D] AutoUpdate: payload: {
13:54:33.114 ->   "current": "0.0.14",
13:54:33.114 ->   "link": "https://github.com/matteocrippa/leafminer/releases/download/v{{version}}/firmware_{{device}}.bin",
13:54:33.160 ->   "devices": [
13:54:33.160 ->     "esp8266",
13:54:33.160 ->     "esp32",
13:54:33.160 ->     "esp32-s2",
13:54:33.160 ->     "esp32-s3",
13:54:33.160 ->     "geekmagic-smalltv",
13:54:33.160 ->     "lilygo-t-display-s3"
13:54:33.160 ->   ]
13:54:33.160 -> }
13:54:33.160 -> 
13:54:33.160 -> [D] AutoUpdate: Remote Version: 0.0.14
13:54:33.160 -> [D] AutoUpdate: New Version: 0.0.14
13:54:33.160 -> [D] AutoUpdate: Device supported: esp8266
13:54:33.160 -> [D] AutoUpdate: Downloading: https://github.com/matteocrippa/leafminer/releases/download/v0.0.14/firmware_esp8266.bin
13:54:34.134 -> [E] AutoUpdate: httpCode: 404
13:54:34.134 -> [I] Network: Connecting to DRE2306...
13:54:36.535 -> [D] Network: Connecting to host public-pool.io...
13:54:37.227 -> [I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.13", null]}
13:54:37.227 -> 
13:54:37.227 -> [I] Network: >>> {"id":2,"method":"mining.authorize","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","x"]}
13:54:37.227 -> 
13:54:37.227 -> [I] Network: >>> {"id":3,"method":"mining.suggest_difficulty","params":[0.000100]}
13:54:37.274 -> 
13:54:37.461 -> [I] Network: <<< [subscribe] {"id":1,"error":null,"result":[[["mining.notify","57ede2bd"]],"57ede2bd",4]}
13:54:37.461 -> [I] Current: New session id: 57ede2bd
13:54:37.649 -> [I] Network: <<< [authorized] {"id":2,"error":null,"result":true}
13:54:37.649 -> [I] Network: Authorized
13:54:37.649 -> [I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.0001]}
13:54:37.649 -> [I] Current: New difficulty: 0.000100000000
13:54:37.649 -> [D] Network: Difficulty set to: 0.0001000000
13:54:37.743 -> [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1b4f578","075252efe6c44319400f9c8fd0343d86fb3fae05000012570000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170305d20c5075626c69632d506f6f6c","ffffffff02e283d2190000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed89162405e8f42ca76a2b7e01018e27c5688c78cb2d37764b1df0628590cbdfa100000000",["64d862a4e53c96e723e6f1820b7ab3e430bf48625cd21b63066e01589f1675a4","f74baf2fc8210c07209b9ce3aad5ceb946310e43743ad83869c1734ee5de43f6","30ea033833f5be872c861bccb05f6ca060449df6e8e6dfa6e7f34e318eec5db3","98eaff5c2c678760c62378d6eeefe824c4288e5f20746fe05acc73e93a6c1560","0039c944166595bd6ad715f824332e8985fa764dd8152e5dfaba416565b26383","5b3540ca2910d164e5e7fd6d810debc8b4ca73660565d8f977c20c7e4a08a41e","f0977fe37842d40d5bf8203b748621f64063bc74b52c5cd2903c56939d98bdf8","26d0025babaecb4862eb9c15b6736d8918fd7b24e89e114f08455f87eca1a333","a4d3489d17488bff0dd23189be417ee9f9864dc2edbac34b165cca970264aca2","7f2c16582934ba843a638afd6f33132622ac8c3074937bb9a494c2fd8330d27b","34264adf108c45e28acd70bf6e92fd6c56b11757fbda44ca120edd5e6efbd355","afcdfda5e92673f36d93ae6c9444684cbc71c171b8d18872b5e95d7c6b62c4ef","5b158048219dfb50d351314640874ac74bfb3046caa0a848d60b28893af5eda3"],"20000000","17034219","6624f093",false]}
13:54:37.836 -> [I] Job: Random value: 638760608
13:54:37.883 -> [I] Current: Job: 1b4f578 ready to be mined
13:54:54.351 -> [D] Current: Hashrate: 21.02 kH/s
13:54:55.402 -> [D] Current: Hashrate: 21.00 kH/s
13:55:06.744 -> [D] Current: Hashrate: 21.02 kH/s
13:55:12.911 -> [D] Current: Hashrate: 21.02 kH/s
13:55:16.273 -> [D] Current: Hashrate: 21.01 kH/s
13:55:23.444 -> [D] Current: Hashrate: 21.02 kH/s
13:55:24.054 -> [I] Miner: [0] > [1b4f578] > 0x000d31d3 - diff 0.000104011930
13:55:24.054 -> [I] Network: >>> {"id":5,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1b4f578","63876060","6624f093","000d31d3"]}
13:55:24.054 -> 
13:55:29.022 -> [I] Current: New highest hashed difficulty: 0.000104011930
13:55:30.287 -> [D] Current: Hashrate: 5.57 kH/s
13:55:33.521 -> [I] Miner: [0] > [1b4f578] > 0x000ea13b - diff 0.000503979439
13:55:33.521 -> [I] Network: >>> {"id":6,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1b4f578","63876060","6624f093","000ea13b"]}
13:55:33.568 -> 
13:55:35.396 -> [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1b4f599","075252efe6c44319400f9c8fd0343d86fb3fae05000012570000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170305d20c5075626c69632d506f6f6c","ffffffff02e283d2190000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed89162405e8f42ca76a2b7e01018e27c5688c78cb2d37764b1df0628590cbdfa100000000",["64d862a4e53c96e723e6f1820b7ab3e430bf48625cd21b63066e01589f1675a4","f74baf2fc8210c07209b9ce3aad5ceb946310e43743ad83869c1734ee5de43f6","30ea033833f5be872c861bccb05f6ca060449df6e8e6dfa6e7f34e318eec5db3","98eaff5c2c678760c62378d6eeefe824c4288e5f20746fe05acc73e93a6c1560","0039c944166595bd6ad715f824332e8985fa764dd8152e5dfaba416565b26383","5b3540ca2910d164e5e7fd6d810debc8b4ca73660565d8f977c20c7e4a08a41e","f0977fe37842d40d5bf8203b748621f64063bc74b52c5cd2903c56939d98bdf8","26d0025babaecb4862eb9c15b6736d8918fd7b24e89e114f08455f87eca1a333","a4d3489d17488bff0dd23189be417ee9f9864dc2edbac34b165cca970264aca2","7f2c16582934ba843a638afd6f33132622ac8c30749 [... cut ...]
13:55:35.537 -> [I] Network: Share accepted
13:55:35.537 -> [I] Current: Hash accepted: 1
13:55:35.631 -> [I] Network: <<< [mining.submit] {"id":6,"error":null,"result":true}
13:55:35.631 -> [I] Network: Share accepted
13:55:35.631 -> [I] Current: Hash accepted: 2
13:55:40.648 -> [I] Current: New highest hashed difficulty: 0.000503979439
13:55:41.351 -> [D] Current: Hashrate: 7.48 kH/s

-------------------------------------

as far as I understood 0.0.14 version, I will observe whether the problem as yesterday stopped just working

tiger0007t, 2024-04-21:
14:01:43.539 -> [I] Job: Random value: 2111886129
14:01:43.539 -> [I] Current: Job: 1c734e8 ready to be mined
14:01:50.272 -> [D] Current: Hashrate: 21.00 kH/s
14:01:54.645 -> [D] Current: Hashrate: 21.01 kH/s
14:01:58.382 -> [D] Current: Hashrate: 21.01 kH/s
14:02:00.446 -> [D] Current: Hashrate: 21.01 kH/s
14:02:01.565 -> [D] Current: Hashrate: 21.01 kH/s
14:02:03.458 -> [D] Current: Hashrate: 21.01 kH/s
14:02:07.053 -> [D] Current: Hashrate: 21.01 kH/s
14:02:09.453 -> [D] Current: Hashrate: 21.01 kH/s
14:02:15.377 -> [I] Miner: [0] > [1c734e8] > 0x000897f0 - diff 0.000338927460
14:02:15.377 -> [I] Network: >>> {"id":5,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1c734e8","21118861","6624f250","000897f0"]}
14:02:15.377 -> 
14:02:15.738 -> [I] Network: <<< [mining.submit] {"id":5,"error":null,"result":true}
14:02:15.738 -> [I] Network: Share accepted
14:02:15.773 -> [I] Current: Hash accepted: 1
14:02:20.751 -> [I] Current: New highest hashed difficulty: 0.000338927460
14:02:25.543 -> [I] Miner: [0] > [1c734e8] > 0x000a2076 - diff 0.000100600298
14:02:25.543 -> [I] Network: >>> {"id":6,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1c734e8","21118861","6624f250","000a2076"]}
14:02:25.543 -> 
14:02:25.822 -> [I] Network: <<< [mining.submit] {"id":6,"error":null,"result":true}
14:02:25.822 -> [I] Network: Share accepted
14:02:25.822 -> [I] Current: Hash accepted: 2
14:02:31.913 -> [D] Current: Hashrate: 11.03 kH/s
14:02:33.737 -> [D] Current: Hashrate: 21.00 kH/s
14:02:36.280 -> [I] Miner: [0] > [1c734e8] > 0x000be071 - diff 0.000156518570
14:02:36.280 -> [I] Network: >>> {"id":7,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1c734e8","21118861","6624f250","000be071"]}
14:02:36.314 -> 
14:02:36.560 -> [I] Network: <<< [mining.submit] {"id":7,"error":null,"result":true}
14:02:36.599 -> [I] Network: Share accepted
14:02:36.599 -> [I] Current: Hash accepted: 3
14:02:45.933 -> [D] Current: Hashrate: 11.91 kH/s
14:02:53.890 -> [D] Current: Hashrate: 21.02 kH/s
14:03:01.154 -> [D] Current: Hashrate: 21.02 kH/s
14:03:09.637 -> [D] Current: Hashrate: 21.02 kH/s
14:03:16.335 -> [I] Miner: [0] > [1c734e8] > 0x001702c9 - diff 0.000198400177
14:03:16.335 -> [I] Network: >>> {"id":8,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1c734e8","21118861","6624f250","001702c9"]}
14:03:16.335 -> 
14:03:17.038 -> [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1c739e7","1b9cd3d16bcfb219ef8ef7af0f886cc001d558950001cd1b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170307d20c5075626c69632d506f6f6c","ffffffff02bf452c1b0000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed9df61fd7872ad9e0f416c27721c7ca4aff755a34eaf318dd2ad45056d5cf196200000000",["4910e09649f1765d2cdf27f9c4b5827624b18aaa58b7d8ab2e067125610dad6c","f4b655b168d5f6b6073569302be62a2cac0bd1e1a8dd430be2d456a93e806310","21cd67e7a5da4d4fe8ecb4f9a2cf487bd5afe2c3400b5602b8e8969ab2821243","1cfb28530bb73de9e6a05bb0854d9ac77d13bb0fb6fad247e8e06f15a6f72d46","4a39819dd7bb8d0b7945757afbee8163e5e6be51561d73caf974036eb2efc32b","50099f521934bcdcdb6f82f08dcd67b24e4e2640ed0c6f62fb83f83f9f903296","26e672248152077bbfa1b4e744f7bb4e256c567419aef532333f4828fe96064b","360f2fa0f8ebc1a7f1c81e77c31fc3966f4eb061d2f169d8df70d46f5da7c79a","8fed99672792092c3a9e26b7c75c19a53ee554b935d476177107c9874a479180","85285bc8e898b13d1293ebc99ee5983044a3f338e8c917b029eba20b5848ab83","c62095295b1a8a3d2fd4c284dabeb688a2807af3ad16f71101d684eaff7e013d","6e5750d533c7b116308d4af17ac574af864cb75d3e3517e7fd3b5cf8be93ee6c","c5c6f9968b900640feacaf859586f8fceed4ec84c79ef5eb0a4def60687fe091"],"20000000","17034219","6624f2b2",false]}
14:03:17.131 -> [I] Job: Random
------------------------------------
then nothing happens, I can't understand why it stops working

tiger0007t, 2024-04-22:
I have a very strange suspicion that nothing works for me as it should due to the fact that platformio compresses the data that needs to be written, this is the first assumption, and the second is that one of my controllers is defined in platformio as an 8285 processor, while writing strangely for memory, and does not have a MAC address defined: ff: ff: ff: ff: ff
-------------------
Uploading .pio\build\esp8266\firmware.bin
esptool.py v3.0
Serial port COM5
Connecting........
Chip is **ESP8285**
Features: WiFi, **Embedded Flash**
Crystal is 26MHz
MAC: ff:ff:ff:ff:ff:ff
-------------------
While the rest of the processors write as ESP8266EX and write their MAC normally
But data compression from platformio:
-------------------
Wrote **650112 bytes (413845 compressed)** at 0x00000000 in 36.4 seconds (effective 142.9 kbit/s).
-------------------
I'll try to program the program now through a web program and see what happens
-------------------
If I program through the site there is no data compression
-------------------
Click Connect to start
Connecting...
Connected successfully.
Try hard reset.
Chip type ESP8266
Connected to ESP8266
MAC Address: D8:F1:5B:11:04:75
Uploading stub...
Running stub...
Stub is now running...
Detecting Flash Size
FlashId: 0x16405E
Flash Manufacturer: 5e
Flash Device: 4016
Auto-detected Flash size: 4MB
Start erasing
Erasing flash memory. Please wait...
Erasing flash memory. Please wait...
Erasing flash memory. Please wait...
Erasing flash memory. Please wait...
Finished. Took 13664ms to erase.
Image header, Magic=0xE9, FlashMode=0x02, FlashSizeFreq=0x40
Writing data with filesize: 650080
Erase size 650080, blocks 40, block size 0x4000, offset 0x0000, encrypted no
Flashing... 2%
Flashing... 5%
Flashing... 100%
Took 61475ms to write 650080 bytes
Erase size 0, blocks 0, block size 0x4000, offset 0x0000, encrypted no
Done!
To run the new firmware please reset your device.

matteocrippa, 2024-04-22:
I'm a bit lost, what's the issue @tiger0007t ?
I see it running, I've just released a v0.0.15 and tested for 1hr with a Weimos D1 and worked fine for me

tiger0007t, 2024-04-23:
probably the log file did not go, I will restart the session and make a log.

tiger0007t, 2024-04-23:
please see log
[Uploading device-monitor-240423-022534.log…]()

tiger0007t, 2024-04-23:
tried to completely overwrite the program, erased the memory completely on ESPWebTool and then compiled it back to platformio, works for 35 minutes without problems at the moment

2:15:39.820 > [I] Network: <<< [mining.submit] {"id":206,"error":null,"result":true}
12:15:39.826 > [I] Network: Share accepted
12:15:39.829 > [I] Current: Hash accepted: 151
12:15:44.821 > [D] Current: Hashrate: 8.92 kH/s
12:15:52.387 > [I] Miner: [0] > [1fb9297] > 0x0081fbdb - diff 0.000842418966
12:15:52.396 > [I] Network: >>> {"id":208,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1fb9297","17136166","66277993","0081fbdb"]}
12:15:52.410 >
12:15:52.621 > [I] Network: <<< [mining.submit] {"id":208,"error":null,"result":true}
12:15:52.627 > [I] Network: Share accepted
12:15:52.630 > [I] Current: Hash accepted: 152
12:15:57.621 > [D] Current: Hashrate: 12.42 kH/s
12:16:01.965 > [I] Miner: [0] > [1fb9297] > 0x00836066 - diff 0.000070748336
12:16:01.974 > [I] Network: >>> {"id":209,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1fb9297","17136166","66277993","00836066"]}
12:16:01.988 >
12:16:02.420 > [I] Network: <<< [mining.submit] {"id":209,"error":null,"result":true}
12:16:02.425 > [I] Network: Share accepted
12:16:02.428 > [I] Current: Hash accepted: 153

tiger0007t, 2024-04-23:
12:22:05.675 > [I] Network: <<< [mining.submit] {"id":246,"error":null,"result":true}
12:22:05.680 > [I] Network: Share accepted
12:22:05.683 > [I] Current: Hash accepted: 176
12:22:10.676 > [D] Current: Hashrate: 10.93 kH/s
12:22:13.914 > [I] Miner: [0] > [1fbe73e] > 0x0009260e - diff 0.000032838167
12:22:13.922 > [I] Network: >>> {"id":247,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1fbe73e","16369603","66277d70","0009260e"]}
12:22:13.935 >
12:22:18.926 > [D] Current: Hashrate: 8.25 kH/s
12:22:24.011 > [I] Miner: [0] > [1fbe73e] > 0x000ac776 - diff 0.000190140712
12:22:24.019 > [I] Network: >>> {"id":248,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1fbe73e","16369603","66277d70","000ac776"]}
12:22:24.032 >
12:24:53.496 > [I] Network: <<< [unknown] {"id":null,"method":"mining.notify","params":["1fbf124","f8e4422d8bd2ece49da62cc80c4acbf82318474f000308890000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff17032ed30c5075626c69632d506f6f6c","ffffffff02dd83e41d0000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9edab8896dae164e9319c7b0b2b419406c96b1004fb11451604b547683bb470f82100000000",["a05043d3511cc2720faff262683c3a6c70eb8c0f6f065662f59f09ec01cd845d","edc245da6dacaf84282607e8d8d37e36ca3759d63e3dee5177
12:24:53.544 > [E] Network: Unknown response type: unknown
12:24:58.538 > [D] Current: Hashrate: 6.12 kH/s
12:25:22.615 > [I] Miner: [0] > [1fbe73e] > 0x002b8909 - diff 0.000035300157
12:25:22.628 > [I] Network: >>> {"id":262,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","1fbe73e","16369603","66277d70","002b8909"]}
-----------------------------------
after a while it ceases to inform about the accepted balls, then swears at the connection

matteocrippa, 2024-04-29:
Can I close this and let's put the esp8266 issue under an unique issue

tiger0007t, 2024-04-30:
of course, it would be convenient to solve this problem in one thread of the topic, and not in different ones, if you can tell me how to get those files out of the repository that I don't have enough to compile code for esp32 and others, devices do not compile from project 3, know how else to synchronize or get these files from the repository, i have esp-01s, esp32, esp12e, also ordered go esp32-s3 for experiments