matteocrippa/leafminer #37: ESP8266 - not visible at pool (no valid shares submitted?)

PSLLSP, 2024-04-23:
ESP8266 running LeafMiner v0.0.14, connected to pool.vkbit.com. Miner is connected, it receives tasks from the pool. I cannot see miner at the pool. I connected miner to PC and I monitor console. I have more than 24 hours of data. Miner reports that it hashes at speed 20 kHps. I cannot see any task that was sent back to the pool and was accepted. I assume there is a bug in LeafMiner and ESP8266 cannot "solve block".

matteocrippa, 2024-04-23:
Try 0.0.15

Just a quick log out of this morning

```log
09:16:21.675 > [I] Miner: [0] > [145c6af] > 0x0009c9c0 - diff 0.000208423532
09:16:21.684 > [I] Network: >>> {"id":6,"method":"mining.submit","params":["xxx.esp8266","145c6af","12456710","66275ff9","0009c9c0"]}
09:16:21.700 > 
09:16:25.141 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145ca21","755532fa562d2f57856d133700b066a29a6e1e11000061600000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031cd30c564b4249542e434f4d","ffffffff033fe724000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3344c1af1c00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed2655870f052b31a820bfbdc0b8dffe10c5b1a66ad41853333c8b404d1a7a763300000000",["bc73946c46d0956928f430fdfa782ac9fe265bf8e1d30b223c9adefe920848cd","bc4bfa54ee1b55e6348f85dc3c8ae34603b8dd33481947ef97d4d6751673d52d","505822c73abc5ebc9f3f06397ca6bdb70ed9dcf8688724f05f3ec488d86976d1","dca927b944d7931850d5f235c8bb47ec605bd95ff674f5c705648f2bcc3897a9","831f872d109f8485930b1b7cc9cda045e54dc82037a143313143cb552553bffc","67bb0e41c61c7f114b39d00401aa5a8662cd41157b47f52ed62bc627a6c169e1","c49f63b395b602803d0a6f699c2ba7e07e97c6f16df087eceb6a155e4bc8d5ff","0591e90e566dbe30592f1e2e90ddf487e504a048d12fe84ccf1b289d42441ee0","bbd2543aef303986f93d4a85350aed508ac9a217fe02d5a44496199bd13ef96e","e1a83fa3a806c56b4169e8791120cbdb2c17985314250956c4b3bdcb2d10033c","f3bcf4b7d068680c2ded03df2b6563b8259769aeaf987909c67ccfc7576cdb10","05c6eaac8771f60dc98fb75e57767ae4a1e3f20a24d226c2a23111d8315d3715","d082dd7015dc5d6e4d35012981370b996a62d371471df576e0025609e88030ae"],"20000000","17034219","66276036",false]}
09:16:25.279 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145cd91","fd8c6a0cd8e538c3f61d5142f0c54600648fe5a0000116d50000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031dd30c564b4249542e434f4d","ffffffff0369fd23000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3383fcf91b00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed3e750c15e44048e533b681b513ff1547bf5ee2fbe1ff9bbbfca075b1af90747a00000000",["6f6c7dafbefc1ae53c69d765731b9b29c04871808cc7901971cc451486623fd9","0a90d2b5d4001bc1ecea03e453fa6a36eb3eb1e8f1bf4e4db6aadbefa576ac7c","be3e1c2bce8c61f48e1e94733a63e2b9205ea1d2227e60ac9ce3d4ce304cca83","fef78b3479b4b31aa132e230637d737c7b0ad5fefa26c1149eb1cb41f588fa01","c58a49d8e710c52559cf2a7a6ddd60f1e82f118a0b136ce9277cb995c2788e2e","6e6bb4427effb89f46c33680981b1b15892e79d9fc5fb8b0808a2de995a2d832","6d315cd599653820032ae3265826017be0db92b5337529827390c756df89e60b","f98054f63f2f5d3af4cce5db12b32bceb7c19c8885776bd6d4cc469bf6694888","75ff959d9920d8bee1a94079ae30fcad81c30d2b9ae6e340a28f49a974324e68","17ed76402bdf394498ff5dd4008e3aa0eb03e4d177c6af0d9e78d42c23f8758d","bb56a2c5811d194ae61ddf46ef3a1f3a0b58c915e4ddfea6f45a27e7d8ceb275","40c47ee8efbd3c76f177f67c43d63073a53b8ad729aae13450ee800c9e78adf1","3184a5235be20bd3eb24013770a241a8145f2a9e290d5d3bddc3f19c6b1bd4c9"],"20000000","17034219","66276039",true]}
09:16:25.402 > [D] Current: Job: 145c6af is cleaned and replaced with 145cd91
09:16:25.407 > [I] Job: Random value: 1094396013
09:16:25.410 > [I] Current: Job: 145cd91 ready to be mined
09:16:25.413 > [I] Network: <<< [mining.submit.fail] {"id":6,"result":null,"error":[21,"Job not found",""]}
09:16:25.421 > [E] Network: Share rejected
09:16:25.424 > [E] Network: Late responses, skip them
09:16:30.418 > [D] Current: Hashrate: 13.95 kH/s
09:16:30.421 > [I] Current: New highest hashed difficulty: 0.000208423532
09:16:44.326 > [I] Miner: [0] > [145cd91] > 0x00047592 - diff 0.000199995053
09:16:44.334 > [I] Network: >>> {"id":9,"method":"mining.submit","params":["xxx.esp8266","145cd91","10943960","66276039","00047592"]}
09:16:44.348 > 
09:16:44.348 > [I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.000016]}
09:16:44.359 > [I] Current: New difficulty: 0.000016000000
09:16:44.361 > [D] Network: Difficulty set to: 0.0000160000
09:16:44.998 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145cda5","fd8c6a0cd8e538c3f61d5142f0c54600648fe5a0000116d50000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031dd30c564b4249542e434f4d","ffffffff0369fd23000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3383fcf91b00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed3e750c15e44048e533b681b513ff1547bf5ee2fbe1ff9bbbfca075b1af90747a00000000",["6f6c7dafbefc1ae53c69d765731b9b29c04871808cc7901971cc451486623fd9","0a90d2b5d4001bc1ecea03e453fa6a36eb3eb1e8f1bf4e4db6aadbefa576ac7c","be3e1c2bce8c61f48e1e94733a63e2b9205ea1d2227e60ac9ce3d4ce304cca83","fef78b3479b4b31aa132e230637d737c7b0ad5fefa26c1149eb1cb41f588fa01","c58a49d8e710c52559cf2a7a6ddd60f1e82f118a0b136ce9277cb995c2788e2e","6e6bb4427effb89f46c33680981b1b15892e79d9fc5fb8b0808a2de995a2d832","6d315cd599653820032ae3265826017be0db92b5337529827390c756df89e60b","f98054f63f2f5d3af4cce5db12b32bceb7c19c8885776bd6d4cc469bf6694888","75ff959d9920d8bee1a94079ae30fcad81c30d2b9ae6e340a28f49a974324e68","17ed76402bdf394498ff5dd4008e3aa0eb03e4d177c6af0d9e78d42c23f8758d","bb56a2c5811d194ae61ddf46ef3a1f3a0b58c915e4ddfea6f45a27e7d8ceb275","40c47ee8efbd3c76f177f67c43d63073a53b8ad729aae13450ee800c9e78adf1","3184a5235be20bd3eb24013770a241a81 [... cut ...]
09:16:52.983 > [I] Current: Hash accepted: 3
09:16:57.975 > [D] Current: Hashrate: 7.21 kH/s
09:16:58.867 > [I] Miner: [0] > [145cda5] > 0x000125e4 - diff 0.000021205302
09:16:58.874 > [I] Network: >>> {"id":12,"method":"mining.submit","params":["xxx.esp8266","145cda5","08899773","66276039","000125e4"]}
09:16:58.889 > 
09:16:59.012 > [I] Network: <<< [mining.submit] {"id":12,"error":null,"result":true}
09:16:59.017 > [I] Network: Share accepted
09:16:59.020 > [I] Current: Hash accepted: 4
```

And I see that running in vkbit.com

[image]

tiger0007t, 2024-04-23:
> Попробуйте 0.0.15
> 
> Просто быстрый выход из этого утра
> 
> ```
> 09:16:21.675 > [I] Miner: [0] > [145c6af] > 0x0009c9c0 - diff 0.000208423532
> 09:16:21.684 > [I] Network: >>> {"id":6,"method":"mining.submit","params":["xxx.esp8266","145c6af","12456710","66275ff9","0009c9c0"]}
> 09:16:21.700 > 
> 09:16:25.141 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145ca21","755532fa562d2f57856d133700b066a29a6e1e11000061600000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031cd30c564b4249542e434f4d","ffffffff033fe724000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3344c1af1c00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed2655870f052b31a820bfbdc0b8dffe10c5b1a66ad41853333c8b404d1a7a763300000000",["bc73946c46d0956928f430fdfa782ac9fe265bf8e1d30b223c9adefe920848cd","bc4bfa54ee1b55e6348f85dc3c8ae34603b8dd33481947ef97d4d6751673d52d","505822c73abc5ebc9f3f06397ca6bdb70ed9dcf8688724f05f3ec488d86976d1","dca927b944d7931850d5f235c8bb47ec605bd95ff674f5c705648f2bcc3897a9","831f872d109f8485930b1b7cc9cda045e54dc82037a143313143cb552553bffc","67bb0e41c61c7f114b39d00401aa5a8662cd41157b47f52ed62bc627a6c169e1","c49f63b395b602803d0a6f699c2ba7e07e97c6f16df087eceb6a155e4bc8d5ff","0591e90e566dbe30592f1e2e90ddf487e504a048d12fe84ccf1b289d42441ee0","bbd2543aef303986f93d4a85350aed508ac9a217fe02d5a44496199bd13ef96e","e1a83fa3a806c56b4169e8791120cbdb2c17985314250956c4b3bdcb2d10033c","f3bcf4b7d068680c2ded03df2b6563b8259769aeaf987909c67ccfc7576cdb10","05c6eaac8771f60dc98fb75e57767ae4a1e3f20a24d226c2a23111d8315d3715","d082dd7015dc5d6e4d35012981370b996a62d371471df576e0025609e88030ae"],"20000000","17034219","66276036",false]}
> 09:16:25.279 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145cd91","fd8c6a0cd8e538c3f61d5142f0c54600648fe5a0000116d50000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031dd30c564b4249542e434f4d","ffffffff0369fd23000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3383fcf91b00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed3e750c15e44048e533b681b513ff1547bf5ee2fbe1ff9bbbfca075b1af90747a00000000",["6f6c7dafbefc1ae53c69d765731b9b29c04871808cc7901971cc451486623fd9","0a90d2b5d4001bc1ecea03e453fa6a36eb3eb1e8f1bf4e4db6aadbefa576ac7c","be3e1c2bce8c61f48e1e94733a63e2b9205ea1d2227e60ac9ce3d4ce304cca83","fef78b3479b4b31aa132e230637d737c7b0ad5fefa26c1149eb1cb41f588fa01","c58a49d8e710c52559cf2a7a6ddd60f1e82f118a0b136ce9277cb995c2788e2e","6e6bb4427effb89f46c33680981b1b15892e79d9fc5fb8b0808a2de995a2d832","6d315cd599653820032ae3265826017be0db92b5337529827390c756df89e60b","f98054f63f2f5d3af4cce5db12b32bceb7c19c8885776bd6d4cc469bf6694888","75ff959d9920d8bee1a94079ae30fcad81c30d2b9ae6e340a28f49a974324e68","17ed76402bdf394498ff5dd4008e3aa0eb03e4d177c6af0d9e78d42c23f8758d","bb56a2c5811d194ae61ddf46ef3a1f3a0b58c915e4ddfea6f45a27e7d8ceb275","40c47ee8efbd3c76f177f67c43d63073a53b8ad729aae13450ee800c9e78adf1","3184a5235be20bd3eb24013770a241a8145f2a9e290d5d3bddc3f19c6b1bd4c9"],"20000000","17034219","66276039",true]}
> 09:16:25.402 > [D] Current: Job: 145c6af is cleaned and replaced with 145cd91
> 09:16:25.407 > [I] Job: Random value: 1094396013
> 09:16:25.410 > [I] Current: Job: 145cd91 ready to be mined
> 09:16:25.413 > [I] Network: <<< [mining.submit.fail] {"id":6,"result":null,"error":[21,"Job not found",""]}
> 09:16:25.421 > [E] Network: Share rejected
> 09:16:25.424 > [E] Network: Late responses, skip them
> 09:16:30.418 > [D] Current: Hashrate: 13.95 kH/s
> 09:16:30.421 > [I] Current: New highest hashed difficulty: 0.000208423532
> 09:16:44.326 > [I] Miner: [0] > [145cd91] > 0x00047592 - diff 0.000199995053
> 09:16:44.334 > [I] Network: >>> {"id":9,"method":"mining.submit","params":["xxx.esp8266","145cd91","10943960","66276039","00047592"]}
> 09:16:44.348 > 
> 09:16:44.348 > [I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.000016]}
> 09:16:44.359 > [I] Current: New difficulty: 0.000016000000
> 09:16:44.361 > [D] Network: Difficulty set to: 0.0000160000
> 09:16:44.998 > [I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["145cda5","fd8c6a0cd8e538c3f61d5142f0c54600648fe5a0000116d50000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff15031dd30c564b4249542e434f4d","ffffffff0369fd23000000000016001427e04fe106a94b902e0f35ce87e0531b0ff47f3383fcf91b00000000160014906d18c86fec38ff10260bdd9ed9e6c2e112fda40000000000000000266a24aa21a9ed3e750c15e44048e533b681b513ff1547bf5ee2fbe1ff9bbbfca075b1af90747a00000000",["6f6c7dafbefc1ae53c69d765731b9b29c04871808cc7901971cc451486623fd9","0a90d2b5d4001bc1ecea03e453fa6a36eb3eb1e8f1bf4e4db6aadbefa576ac7c","be3e1c2bce8c61f48e1e94733a63e2b9205ea1d2227e60ac9ce3d4ce304cca83","fef78b3479b4b31aa132e230637d737c7b0ad5fefa26c1149eb1cb41f588fa01","c58a49d8e710c52559cf2a7a6ddd60f1e82f118a0b136ce9277cb995c2788e2e","6e6bb4427effb89f46c33680981b1b15892e79d9fc5fb8b0808a2de995a2d832","6d315cd599653820032ae3265826017be0db92b5337529827390c756df89e60b","f98054f63f2f5d3af4cce5db12b32bceb7c19c8885776bd6d4cc469bf6694888","75ff959d9920d8bee1a94079ae30fcad81c30d2b9ae6e340a28f49a974324e68","17ed76402bdf394498ff5dd4008e3aa0eb03e4d177c6af0d9e78d42c23f8758d","bb56a2c5811d194ae61ddf46ef3a1f3a0b58c915e4ddfea6f45a27e7d8ceb275","40c47ee8efbd3c76f177f67c43d63073a53b8ad729aae134 [... cut ...] Zhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDQyM1QwODA3MTJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xYmVlNTgxNTRlMDZiYjc0ZjhmODBhOGI0YjI2OGZiYTZjYTVjMGJmZDBiMTg0YTY4MjNhYzZiZjMyMzQ2NTAzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.pe-l5-x8xiRcff9EOHaeRfeYvJMpUG5-x2S8vUmzUE0)

Yesterday, after the crash, I tested the 0.0.15 version, posted the log file of the work in the next branch, there are both brute force errors and only the hash goes and a more serious error.

PSLLSP, 2024-04-23:
I see this with ESP8266 with 0.0.14:

```
...
[I] Miner: [0] > [12e60ef] > 0x0016fdfb - diff 0.002970821552
[I] Network: >>> {"id":142,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","0016fdfb"]}

[I] Current: New highest hashed difficulty: 0.002970821552
[I] Miner: [0] > [12e60ef] > 0x0017aa2a - diff 0.000052325939
[I] Network: >>> {"id":143,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","0017aa2a"]}

[I] Miner: [0] > [12e60ef] > 0x00192873 - diff 0.000060454863
[I] Network: >>> {"id":144,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","00192873"]}

[I] Miner: [0] > [12e60ef] > 0x0019bf60 - diff 0.000016961303
[I] Network: >>> {"id":145,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","0019bf60"]}

[I] Miner: [0] > [12e60ef] > 0x001a8674 - diff 0.000020856352
[I] Network: >>> {"id":146,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","001a8674"]}

[I] Network: <<< [mining.submit] {"id":142,"error":null,"result":true}
[I] Network: Share accepted
...
[I] Miner: [0] > [12e60ef] > 0x613c9895 - diff 0.000035406287
[I] Network: >>> {"id":23883,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","613c9895"]}

[I] Miner: [0] > [12e60ef] > 0x613cdc7e - diff 0.000063324575
[I] Network: >>> {"id":23884,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","613cdc7e"]}

[I] Miner: [0] > [12e60ef] > 0x613d3d34 - diff 0.004096409945
[I] Network: >>> {"id":23885,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","613d3d34"]}

[I] Miner: [0] > [12e60ef] > 0x613dea7c - diff 0.000016730604
[I] Network: >>> {"id":23886,"method":"mining.submit","params":["xxx.LEAF8266","12e60ef","65035258","662614bd","613dea7c"]}
```

Result 142 was the last one that was confirmed from the pool... Well, it is the last message received, after that I see only TX messages...

tiger0007t, 2024-04-23:
I also had such a problem

tiger0007t, 2024-04-28:
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\esp32\src\miner\miner.cpp.o
Compiling .pio\build\esp32\src\miner\nerdSHA256plus.cpp.o
*** [.pio\build\esp32\src\current.cpp.o] Error 1
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32\src\main.cpp.o] Error 1
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32\src\miner\miner.cpp.o] Error 1
xtensa-esp32-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32\src\miner\nerdSHA256plus.cpp.o] Error 1
--------------------------------
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\esp32-s3\src\miner\miner.cpp.o
*** [.pio\build\esp32-s3\src\current.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32-s3\src\main.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\esp32-s3\src\miner\miner.cpp.o] Error 1
--------------------------------
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
Compiling .pio\build\lilygo-t-display-s3\src\miner\nerdSHA256plus.cpp.o
*** [.pio\build\lilygo-t-display-s3\src\current.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\lilygo-t-display-s3\src\main.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\lilygo-t-display-s3\src\miner\miner.cpp.o] Error 1
xtensa-esp32s3-elf-g++: error: CreateProcess: No such file or directory
*** [.pio\build\lilygo-t-display-s3\src\miner\nerdSHA256plus.cpp.o] Error 1
--------------------------------
Environment          Status    Duration
-------------------  --------  ------------
esp8266                     SUCCESS   00:00:01.537
geekmagic-smalltv    SUCCESS   00:00:01.738
esp32                         FAILED    00:00:01.586
esp32-s2                    SUCCESS   00:00:04.321
esp32-s3                    FAILED    00:00:01.714
lilygo-t-display-s3     FAILED    00:00:01.965
===3 failed, 3 succeeded in 00:00:12.860 ===

can't assemble a project, on esp32, esp32-s3, lilygo-t-display-s3, swears at files, synchronization does not work with github. A snippet of code for what curses, how to add or synchronize branches?

matteocrippa, 2024-04-28:
Try to clean the project that seems the cause, but if you have just an esp8266 you should just build and upload that one not all

tiger0007t, 2024-04-28:
cleaned up the project completely. downloaded the new version 0.0.16 test, worked for 25 minutes and again went into error

[D] Network: <<< len: 37

[I] Network: <<< [mining.submit] {"id":147,"error":null,"result":true}

[I] Network: Share accepted

[I] Current: Hash accepted: 108

[D] Network: <<< len: 0

[D] Current: Hashrate: 7.44 kH/s

[D] Current: Hashrate: 21.00 kH/s

[D] Miner: [0] > Hash 0.000041856047 > 0.000032000000

[I] Miner: [0] > [273fe90] > 0x00111276 - diff 0.000041856047

[I] Network: >>> {"id":148,"method":"mining.submit","params":["bc1qgjlz0dw8aq0pq3sradky8jtqy8mrwfyn2z2c0z","273fe90","15609009","662eb76d","00111276"]}

[D] Network: <<< len: 1128

[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["2740da0","595425bb6bb21985c90b19cf9f0914190445a43500024a350000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170345d60c5075626c69632d506f6f6c","ffffffff025d5ed7130000000016001444be27b5c7e81e104603eb6c43c96021f63724930000000000000000266a24aa21a9ed595f8ef66e4bd120bc6a36b970d342fa729c34b893e40a23a5e1928d29e971c600000000",["0c438b279a41a627980c274aa843a9e18ce44d5932fe6661c952e4590524c617","d1e30fac64698168d1aecdc4b3c6113dfecab9f8c8a8652fd7687346ca29bfff","a18806cdaae9f27aed3294a77521e3e146c36345fa430fa00091c03d49ad04ce","3c0484e0b4fec4492bc17f0a3ab516374a0e32b98ef47fa53af30d93b75516c5","8a098ca930bb852897cb7936c72008e396b810b72c2c63b2b23440b8ce1db6d3","344e37a1fe4cfe1f61237fe09ac1433d30ec1ea4d6937756a639e83720d45960","28d5d85d03f43dbd80b1f10cb3c57f0bd3beb9f9866130cb0ea5491aa55cf76b","7ed7bb85af626c6f11db4f9e8036dd4f7dea4e764abe269307886b39cfd00da2","6c9d3494719faede43e0e3ec9c6a80f8d27033c36685634895879c109c60c0e9","df0dd05dc5c4c4252ae149176dce28305e2fc1bd6a95028ec173d4f36a13feee"],"20000000","170331db","662eb7da",true]}

[D] Current: Job: 273fe90 is cleaned and replaced with 2740da0

[I] Job: Random value: 1544622183

[I] Current: Job: 2740da0 ready to be mined

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:191 __yield

>>>stack>>>

ctx: cont
sp: 3ffff560 end: e2f156c4 offset: 0010
3ffff580:  000007ff 3ffff5e8 3fff12b0 40226851  
3ffff590:  000012b3 00000020 00000000 40226955  
3ffff5a0:  3fff12b0 00000020 3fffb2dc 3fff416c  
3ffff5b0:  3fff4184 00000020 3fffb2dc 3fff416c  
3ffff5c0:  3fff4184 3fff419c 00000468 402157f2  
3ffff5d0:  3fffb2dc 00000000 00000468 ffffffff

and so on

matteocrippa, 2024-04-28:
But is later stuck or auto-restarts?

tiger0007t, 2024-04-29:
I will send a log file, it does not restart and goes on a loop
[device-monitor-240428-232926.log](https://github.com/matteocrippa/leafminer/files/15145782/device-monitor-240428-232926.log)
tell me how to fix the problem with synchronization when compiling for other processors, esp32, esp32-s3, lilygo-t-display-s3 I completely reinstalled version 16 in platformio if I'm not mistaken, the system itself must load data from the github site, maybe the repository does not have these files on github?

matteocrippa, 2024-04-29:
@PSLLSP is now working for you and are you able to see the shares?

matteocrippa, 2024-04-29:
@tiger0007t use the binary for now, I think you need to pull the changes from the repo

ArakelTheDragon, 2024-07-21:
The miner is not recognized by the pool at all. I am using the sha256 pool.
Also in which file are the errors with the description, since I can not find it?

Can someone answer what is wrong with this hash:
[I] Network: <<< [mining.submit] {"id":54,"error":[25,"Not subscribed",null],"result":null}

[code]
[I] Network: Connected to WiFi
[I] Network: IP address: 192.168.1.128
[I] Network: MAC address: C4:D8:D5:01:04:89
[D] Network: Connecting to host sha256.unmineable.com...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.16", null]}

[I] Network: >>> {"id":2,"method":"mining.authorize","params":["CAKE:0xFBd767f6454bCd07c959da2E48fD429531A1323A.ESP#cytf-7whi","x"]}

[I] Network: >>> {"id":3,"method":"mining.suggest_difficulty","params":[0.000100]}

[D] Network: <<< len: 139
[I] Network: <<< [subscribe] {"id": 1, "result": [[["mining.set_difficulty", "58ada53cb6cb7579"], ["mining.notify", "58ada53cb6cb7579"]], "f1f4c7cc", 8], "error": null}
[I] Current: New session id: 58ada53cb6cb7579
[D] Network: <<< len: 40
[I] Network: <<< [authorized] {"id": 2, "result": true, "error": null}
[I] Network: Authorized
[D] Network: <<< len: 66
[I] Network: <<< [mining.set_difficulty] {"id": null, "method": "mining.set_difficulty", "params": [16384]}
[I] Current: New difficulty: 16384.000000000000
[D] Network: Difficulty set to: 16384.0000000000
[D] Network: <<< len: 1595
[I] Network: <<< [mining.notify] {"id": null, "method": "mining.notify", "params": ["30b6", "980eec7bbbf943db011ad9087fac3353e99c774d0001c5690000000000000000", "01000000010000000000000000000000000000000000000000000000000000000000000000ffffffff5e0325050d172f5669614254432f4d696e65642062792065746230782f2cfabe6d6de127748bd49b94e4c2bea5691d7bc2a2ebbfa26ac0b3ee77f60405378b05dd95100000000000000010b63091011d291625", "ffffffff0378b2e612000000001976a914fb37342f6275b13936799def06f2eb4c0f20151588ac00000000000000002b6a2952534b424c4f434b3a3846ad00098c074c6099d7f2f57fd83376decf2bec618f551f1784250063d1440000000000000000266a24aa21a9edc4bd0293df18cdf5ecab9eaa70156a7aa82ef64df3b9bd8f592301c9781b20f100000000", ["3ccb6ac55403a3c52ceac71fe898e12fd418ef116167a1c0463f1b09546ec361", "4e520bd2c6e4ba754337082dde0c287abb8eee08f93aa3cbb250a2792b2be8aa", "d08617f42fae9da2c5ce5b4bd5a31a297ba116878bb23213d4fcb2b1155d64d1", "3711b427c4786bdb9615a4dc0e9457793deb2f27e794ab4b63fa1649b981af1c", "7e6705982bcfc3aced0b0aeb94477c25f1df171ee5cdf9bdff9314dfeca50dc9", "5600ea97ced14a241fdc35b127fca88e2508ab9bf994a521fee9b963965120c3", "d6dc0935d57c8b4a3ccb87c22ba84d00db25859ef9cac2859557ac7bc8b45745", "c5825573522dab2afd4c46572c662b63ad105dbd42030755e1777f410d58c40c", "e36c62bc290aa8ecc7b8166686ce447b0db2dab579957420bbdd79b423242e1a", "1ca8384a2997c4d52b1e7dc600dc3829e07f6a3da4de101740e59f2444b79139", "1d1845810f1172120c98c15978828e15b9a3edb60016863cca79d044315cfe12", "10f4a01bd22074ea5a85eb7e6e6e783fb0f1ef6247b60793250f44bce7eda150", "636b4404c4a8def054c2e566a87723d1f3c0dcbd804f3943052c6a4746b3a520"], "20000000", "17036e3a", "669d8f80", true]}
[I] Job: Random value: 228383541
[I] Current: Job: 30b6 ready to be mined
[D] Network: <<< len: 40
[I] Network: <<< [mining.submit] {"id": 3, "error": null, "result": true}
[I] Network: Share accepted
[I] Current: Hash accepted: 1
[D] Network: <<< len: 62
[I] Network: <<< [mining.set_difficulty] {"id": null, "method": "mining.set_difficulty", "params": [0]}
[I] Current: New difficulty: 0.000000000000
[D] Network: Difficulty set to: 0.0000000000
[/code]

PSLLSP, 2024-07-22:
> Also in which file are the errors with the description, since I can not find it?

That is an output from serial console...

matteocrippa, 2024-07-22:
Can be the pool is not accepting low difficult miners? I recall some do not consider them if below some limit, consider that an ESP32/8266 has a ridiculous max difficulty

PSLLSP, 2024-07-22:
You try to mine at Unmineable. It was possible in the past with NerdMiner, some details are documented in [NerdMinerV2 - issue 219](https://github.com/BitMaker-hub/NerdMiner_v2/issues/219)

There was a bug in unmineable pool, it was possible to mine there with NerdMiner but it is possible that they already fixed the bug and NerdMiner cannot generate any valid shares anymore...

Anyway, I do not think it is good idea to mine with LeafMiner at Unmineable, because Unmineable doesn't support solo mining and hashing power of LeafMiner is so low, that it cannot be profitable; it is a solo miner and you play game all or nothing... ;-)

ArakelTheDragon, 2024-07-22:
I am looking for something like 500H/s max if possible, it can be with xmrig, sha256, lyrav2, scrypt, ghostrider or any of the miners, currently my PC gets 1kH/s, even is the nodeMCU can get 50H/s, its still good for me. I have low difficulty pools with 250 difficulty.

matteocrippa, 2024-07-22:
250 is way too high for an esp32

ArakelTheDragon, 2024-07-22:
I can build the circuit for a couple of PIC18F3550 CPUS, each one is 96MHz with the PLL, and interface it with an ESP8266 or ESP32 for the WiFi. But the coding for this will be complex, will it be possible to use the PICs to do the hashes and the ESP for the WiFi communication?