matteocrippa/leafminer #39: Support for ESP8266 (ESP01)

ffrediani, 2024-04-29:
Hello

Is ESP01 (https://www.theengineeringprojects.com/wp-content/uploads/2019/03/introduction-to-esp-01.jpg) supported ?
I managed to write the binary to an ESP01 but then when it boots nothing happens. It doesn't show the LEAFMINER SSID. As I understand it has 1MB of flash and the image size is 638kB so it fits.

Thanks

matteocrippa, 2024-04-29:
I don't have that board, probably some changes are needed and some feature flags needs to be introduced for that.
I tag it as enhancement for now

TOMASLGA, 2024-05-02:
I,ve the same problem.

TOMASLGA, 2024-05-04:
For de ESP-01S I,ve changed the line with the tipe of board esp8266 in the platformio.ini file "board = esp01_1m" and it works, in my case not for long time, but it appears at the pool page and I could configure it by wifi at least. I need to reset it many times to see it working again because it stops. I,m in the way.

ffrediani, 2024-05-04:
Hi @TOMASLGA 
Thanks for sharing it. I will validate for my ESP01. However you mentioned ESP-01S and mine as ESP01. Doing some search here it looks like there are 3 variants: 1) Blue ESP-01 with 512kB of flash, 2) Black ESP-01 and Black ESP-01S both with 1M of flash. Maybe that one I've tested before was the Blue ESP-01 with 512kB of flash which would not have space for Leafminer image which is 638kB.

Both use the same chip, the main difference between ESP01 and ESP8266-NodeMCU 1.0 which may use ESP12E are the number of pins available and 4MB of flash which I guess for Leafminer propose doesn't make any practical difference.

Regarding the time it mines until stops I guess it may have to do with issue #24 because it keeps happening with ESP8266 DevKit/NodeMCU which it still to be worked out, but worth to find out if there are any specific issues to ESP01.

TOMASLGA, 2024-05-04:
Hi @ffrediani , yes my ones are esp01s

[image]

There you can see how it works at the pool.

This is the size of the file when you write it:
[image]

And for yout board I,ll change for esp01.

[image]

dwphilbrick, 2024-06-03:
Is there a way to add support for the OLED 128x64 screen? I have one connected with my ESP8266, but it currently stays black while running the current version.

ffrediani, 2024-06-03:
@dwphilbrick are you talking about ESP01 specifically or ESP8266 (another model) ?

ffrediani, 2024-06-05:
@TOMASLGA it didn't work for me just by changing "board = esp01_1m", "board_build.f_cpu = 80000000L" and adding "board_build.cu = esp8266". At the end it seems to compile but when flashed to ESP01 after reboot nothing happens. Could you detail it a little bit more ?

matteocrippa, 2024-06-05:
feel free to submit a PR with that!