matteocrippa/leafminer #40: Reconfigure WiFi / Pool address / Wallet

ffrediani, 2024-05-04:
As after the first configuration and reboot the web interface stops responding it is useful sometimes to be able to reconfigure the board to connect to another Wifi network, another pool or change the wallet address. It is understandable the web server may use valuable resources that can impact on the hashrate. A few ways I see this can be achieve are:

1 - If simply leaving the web server enabled doesn't use any extra CPU resources when there is no opened connections it can be left with a simple form page that doesn't use CPU resources (like to show updated hashrate or other details) and through the web interface be able to change these details, save and reboot.

2 - If that is not possible, than through the pressing of the flash/boot bottom stop the mining job and start the web server. In order to give feedback to the user blink the ESP light in a certain sequence to confirm the change, and keep blinking in a different way to show that it is not mining and waiting with the web server running.

3 - If the board is taken to another place where the SSID previously configured cannot connect, the ESP notice that after some time, configure itself in AP mode and start the web server for a new configuration.

4 - If none of the above is worth implementing, simply if the flash/boot button is pressed for a few seconds while it is running (not before power on) the ESP resets its configuration and reboot in the default firmware configuration. In the case of ESP8266 Devkit can't the flash/boot button be used for this as well ?

matteocrippa, 2024-05-04:
1. Sadly will involve to have the web server running and will kill the already poor hash rate
2. On devices with 2 buttons, is enough pressing to `For ESP32: Unplug the USB cable, press and hold the BOOT button, and then plug in the USB cable.` and this will force the setup again
3. This should be in place too, if WiFi does not connect it reset the AP and the setup system
4. Yes, it's what we are doing https://github.com/matteocrippa/leafminer?tab=readme-ov-file#resetting-setup

ffrediani, 2024-05-04:
Hi Mateo
ESP8266-NodeMCU Devkit has 2 buttons where the flash/boot one can may be used too to avoid the need to re-flash.
Great about point 3 then.

For 4 not sure I understood correctly. What is said on the URL is that ESP8266 has a limitation (for when there is only a single button). Where there are 2 is it already something possible or needs implementation ? The idea is to when pressing it for a few seconds *while it is running* it resets its current configuration to factory and reboots. I need to test, but I understand that if pressing the flash button before power on the device will not boot and will instead go into a state where it awaits a new flash, or am I confused ?

matteocrippa, 2024-05-04:
You can see it here https://github.com/matteocrippa/leafminer/blob/2db6981ff3d5af11ae3cb74e1f714f5bf3817917/src/main.cpp#L59

Basically I check if the button is pressed on start and if so I force the Access Point once again.
For now for esp8266 there's no other way to wipe it

ffrediani, 2024-05-04:
Thanks Mateo
Would it perhaps be a option for 8266 when the button is pressed to stop the mining job and bring up the Access Point for reconfiguration ?

matteocrippa, 2024-05-04:
I recall the single button on esp8266 it's sadly not scriptable because it's just force the reboot

ffrediani, 2024-05-04:
Hi Mateo
Not sure we are talking the same board, but at least on the NodeMCU one there are 2 buttons, where one is the RST one that does exactlly what you said, but there is another one called Flash that perhaps can be used after boot. See the picture below:

[image]

matteocrippa, 2024-05-04:
Ah ok, so we can add its GPIO and handle that.
I'm actually using some Weimos D1, and those have just a single RST button

matteocrippa, 2024-05-06:
@ffrediani do you know by any chance the GPIO for that button?

ffrediani, 2024-05-06:
Hello @matteocrippa 
It is connected to GPIO 0 (or D3)