matteocrippa/leafminer #41: Add IPv6 support to connect to pools

ffrediani, 2024-05-04:
Can you add IPv6 TCP/IP stack support to Leafminer along with IPv4 in order that miners can connect to pools preferably via IPv6 rather than IPv4. This avoids having passing via a connection with NAT or even ISP CGNAT and can favor a bit more privacy in certain scenarios.
Although still a lot of people forget, IPv6 is responsible for over 50% of the Internet traffic on some networks given the biggest content generators worldwide fully support IPv6. IPv6 is the actual Internet Protocol and IPv4 is legacy already.
There are already several VPS and hosting offerings IPv6 as well.

Normally most modern operating systems with dual-stack (IPv4 and IPv6) support try to automatically connect preferably via IPv6 if both are available at source and destination. The important thing is that an IPv6 address can be received or autoconfigured by the board and when connecting to pool it will resolve the destination FQDN to IPv6 and use it to connect.
The simplest way the board can get an IPv6 address from the router is via RA/SLAAC (Stateless) which is default for almost any router with IPv6 support.

Both ESP32 and ESP9266 have support for a while in the LWIP since RTOS SDK v1.3.0
For reference Tasmota firmware has IPv6 support for both boards (https://tasmota.github.io/docs/IPv6/). For ESP32 is default, for ESP8266 requires a specific build probably due to the image size which may not be a problem for Leafminer which seems to have a smaller footprint and binary size, but anyway it says it should use an extra 20-30kB extra flash size.
In their documentation it mentions "SetOption149 1"in order to query AAAA before A.

For the most know Solo pools the following have full IPv6 support:
 - public-pool.io
 - pool.vkbit.com
 - solo.ckpool.org
 - btc.solomining.io

References:
https://tasmota.github.io/docs/IPv6/
https://github.com/esphome/esphome/pull/5449
https://esphome.io/changelog/2024.3.0.html#ipv6

matteocrippa, 2024-05-04:
It's a good improvement, for now I will put on hold till we don't have a stable mining on esp8266

ffrediani, 2024-05-04:
Thanks Matteo
Perhaps something can be tried for ESP32 at least for now.