matteocrippa/leafminer #42: discrepency in setup steps

nickfromthebus, 2024-05-05:
in setup steps it says to use address in filename for bin files. In example of esp32-s3 however the example given is

0x0000  bootloader.bin
0x8000 partitions.bin
0xE000
0x10000 firmware.bin

however the filename for s3 version of bootloader is 0x1000_bootloader.bin and there is an 0xE000_boot.bin which is left blank. So I am confused as to proper addressing. Are the files labeled incorrectly or is the example wrong? And what is with the missing E000 address in example whereas the file exists? I am asking because I cannot get my ESP32 to work properly but the example is VERY unclear

[image]
[image]

matteocrippa, 2024-05-06:
@nickfromthebus the readme needs some updates and will be address in the next PR, btw feel free to refer to the structure in the archive, all the files for esp32-s3 are needed, and you can spot them ending with esp32-s3.

If you are installing from scratch you will need to install:
- 0x1000_bootloader_esp32-s3.bin
- 0xe000_boot_esp32-s3.bin
- 0x10000_firmware_esp32-s3.bin
- 0x8000_partition_esp32-s3.bin

Then any further upgrade will need just the firmware-esp32-s3.bin

8lackshiki, 2024-06-30:
I am trying to install in ESP8266 MOD (CH340C driver) and am confused as well with the above.

Please suggest a correct file name for the ESP8266:
- 0x1000_?
- 0xe000_?
- 0x10000_?
- 0x8000_?

Thank you

matteocrippa, 2024-06-30:
@8lackshiki download the files ending with esp8266 from here and in front there's the address to each of them