matteocrippa/leafminer #43: Can't we change the digging logic and do daily digging?

lololololo42, 2024-05-07:
have a nice day
I follow your project with interest. Unfortunately, there are minor problems in Esp 32 and 8266, I hope they will be fixed. Anyway, the goal here is to simulate a small miner, a bit more like a pickaxe. As Kazim, there is no change. Can't we earn even if the daily rate is 0.0000000000000001 BTC? What trade-offs would we need to make for an alternative currency? As far as I understand, nerdminer changed its infrastructure and built it.

matteocrippa, 2024-05-07:
The project has targeted bitcoin solo mining, as you know by definition it's a lotto game and statistics does not help us.
The idea is to get familiar with ESP32/8266 and mining.
You can even consider alternate currencies, it's a matter to add the needed logic and adjust the stratum, but I'm not sure any will be profitable working with this CPU power

lololololo42, 2024-05-08:
Thanks for the answer.
As I said, I want to learn how to use ESP32/8266 with devices such as ardino pico. Small devices are very good, the goal is not financial gain but to understand their functioning. There is Duino, for example, but I cannot mine because the memberships are not created as a closed wallet. For example, +0.0000000001 ETH as soon as I started mining. In fact, as I wrote yesterday, even if it is 1 BTC per billion, the effect is different. Imagine, a tiny device releases 0.000000000000001 BTC per day. This is an interesting and strange hobby :D How should I go about this? Thank you.