matteocrippa/leafminer #48: I can't see my ESP8266 worker at pool.vkbit.com or public-pool.io

prosselloe, 2024-08-13:
Good afternoon.

I'm a technologies teacher and I want to introduce my students in Bitcoin mining.

In LeafMiner you suggest pool.vkbit.com, but when configured, I can't see my ESP8266 worker at it!
I tried also with public-pool.io, but I have the same problem. 
I only can view my device at pool.nerdminers.org, but I prefer the last one.

Do you know where can be the problem? What can I do?

Thanks a lot.

prosselloe, 2024-08-13:
Related issues:

- [Change default pool to public-pool.io](https://github.com/matteocrippa/leafminer/issues/44)
- [ESP8266 - not visible at pool (no valid shares submitted?)](https://github.com/matteocrippa/leafminer/issues/37)

matteocrippa, 2024-08-13:
@prosselloe Happy to hear about introducing students to Bitcoin mining, the docs needs for sure a revamp as I recall the vbkit.com is not anymore supporting low difficulty.

For public-pool.io it should work if nothing outstanding has changed, at the end do suppose it uses the same version of pool.nerdminers.org.

Do you have by any chance any log out of public-pool.io?

I have no esp(s) next to me in this moment so it's a bit tricky to test it

prosselloe, 2024-08-16:
Here you have an examlple:

[D] Network: <<< len: 0
[I] Network: Connecting to sagemcomFE20...
[I] Network: Connecte[D] Current: Hashrate: 21.01 kH/s
[D] Miner: [0] > Hash 0.000852483860 > 0.000100000000
[I] Miner: [0] > [1734bbe] > 0x06b4b227 - diff 0.000852483860
[I] Network: >>> {"id":241,"method":"mining.submit","params":["3NH6hkTmy7WcAL37LaUCbyZX1cDM4jXVzq.ESP-AE7DB6","1734bbe","85415904","66bf41c3","06b4b227"]}

[D] Network: <<< len: 0
[I] Network: Connecting to sagemcomFE20...
[I] Network: Connected to WiFi
[I] Network: IP address: (IP unset)
[I] Network: MAC address: 84:CC:A8:AE:7D:B6
[D] Network: Connecting to host public-pool.io...
[D] Network: Connecting to host public-pool.io...
[D] Miner: [0] > Hash 0.000118492436 > 0.000100000000
[I] Miner: [0] > [1734bbe] > 0x06b5ac4c - diff 0.000118492436
[I] Network: >>> {"id":242,"method":"mining.submit","params":["3NH6hkTmy7WcAL37LaUCbyZX1cDM4jXVzq.ESP-AE7DB6","1734bbe","85415904","66bf41c3","06b5ac4c"]}

[D] Network: <<< len: 0
[I] Network: Connecting to sagemcomFE20...
[I] Network: Connected to WiFi
[I] Network: IP address: 192.168.0.18
[I] Network: MAC address: 84:CC:A8:AE:7D:B6
[D] Network: Connecting to host public-pool.io...
[D] Current: Hashrate: 5.06 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 20.99 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 20.99 kH/s
[D] Current: Hashrate: 21.01 kH/s
[D] Miner: [0] > Hash 0.000113918082 > 0.000100000000
[I] Miner: [0] > [1734bbe] > 0x06baee1d - diff 0.000113918082
[I] Network: >>> {"id":243,"method":"mining.submit","params":["3NH6hkTmy7WcAL37LaUCbyZX1cDM4jXVzq.ESP-AE7DB6","1734bbe","85415904","66bf41c3","06baee1d"]}

[D] Network: <<< len: 0
[I] Network: Connecting to sagemcomFE20...
[I] Network: Connected to WiFi
[I] Network: IP address: 192.168.0.18
[I] Network: MAC address: 84:CC:A8:AE:7D:B6
[D] Network: Connecting to host public-pool.io...
[D] Current: Hashrate: 3.44 kH/s
[D] Miner: [0] > Hash 0.000291851949 > 0.000100000000
[I] Miner: [0] > [1734bbe] > 0x06bd3759 - diff 0.000291851949
[I] Network: >>> {"id":244,"method":"mining.submit","params":["3NH6hkTmy7WcAL37LaUCbyZX1cDM4jXVzq.ESP-AE7DB6","1734bbe","85415904","66bf41c3","06bd3759"]}

[D] Network: <<< len: 0
[I] Network: Connecting to sagemcomFE20...
[I] Network: Connected to WiFi
[I] Network: IP address: 192.168.0.18
[I] Network: MAC address: 84:CC:A8:AE:7D:B6
[D] Network: Connecting to host public-pool.io...
[D] Current: Hashrate: 12.73 kH/s
[D] Current: Hashrate: 21.01 kH/s
[D] Miner: [0] > Hash 0.000146904586 > 0.000100000000
[I] Miner: [0] > [1734bbe] > 0x06c2de57 - diff 0.000146904586
[I] Network: >>> {"id":245,"method":"mining.submit","params":["3NH6hkTmy7WcAL37LaUCbyZX1cDM4jXVzq.ESP-AE7DB6","1734bbe","85415904","66bf41c3","06c2de57"]}

Best regatds.

matteocrippa, 2024-08-16:
It's a bit strange it continues to reconnect after submit, a this point something is happening in the response.
Need to check as soon I have a device around, but not later than next week

g33xt4h, 2024-10-01:
Hi, i have the same issue. Tried it with 0.16 and 0.17 - here are the results with an esp8266 12-F and 0.17:

Stack smashing detected.

Exception (5):
epc1=0x401003fa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffff90 end: 3fffffb0 offset: 0000
3fffff90:  3fffdad0 00000000 3fffdab0 401003fa  
3fffffa0:  feefeffe 00000000 3fffdab0 40229f2c  
<<<stack<<<

matteocrippa, 2024-10-02:
I just merged the v/0.0.17 branch - https://github.com/matteocrippa/leafminer/pull/50
Let me know if it works anyhow better

g33xt4h, 2024-10-06:
Thanks for the merge, i tested it again and sadly it did crash again.
Here are the results:

[I] Current: Job: 1857787 ready to be mined
[D] Network: <<< len: 0

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Stack smashing detected.

Exception (5):
epc1=0x401003fa epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffff90 end: 3fffffb0 offset: 0000
3fffff90:  3fffdad0 00000000 3fffdab0 401003fa  
3fffffa0:  feefeffe 00000000 3fffdab0 40229f2c  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)
load 0x4010f000, len 3424, room 16 
tail 0
chksum 0x2e
load 0x3fff20b8, len 40, room 8 
tail 0
chksum 0x2b
csum 0x2b
v0009fe10
~ld
����o��r��o|�$��l "���{�$�N��o�$`��r�p��l��[I] Main: LeafMiner - v.0.0.17 - (C: 1)
[I] Main: Compiled: Oct  6 2024 19:28:56
[I] Main: Free memory: 36136

matteocrippa, 2024-10-06:
But does it happen as soon as the first job arrives or after a few time / jobs?

g33xt4h, 2024-10-07:
It's everytime as soon as the second job is about to begin. First job is being transmitted ok.
This time the esp just hangs without a crash, so no exception code.

Seems a bit random if it crashes or hangs.

`[I] Main: LeafMiner - v.0.0.17 - (C: 1)
[I] Main: Compiled: Oct  6 2024 19:28:56
[I] Main: Free memory: 36136
[I] Main: ESP8266 - Disable WDT
[I] Storage: Setup: OK
[I] Configuration: wifi_ssid: XXX
[I] Configuration: wifi_password: XXX
[I] Configuration: wallet_address: XXX.esp1
[I] Configuration: pool_password: 
[I] Configuration: pool_url: public-pool.io
[I] Configuration: pool_port: 1073741680
[I] Configuration: blink_enabled: off
[I] Configuration: blink_brightness: 256
[I] Configuration: lcd_on_start: on
[I] Configuration: miner_type: 
[I] Configuration: auto_update: off
[I] Blink: Blinking 5 times
[I] Network: Connecting to XXX...
[I] Network: Connected to WiFi
[I] Network: IP address: X.X.X.X
[I] Network: MAC address: X:X:X:X:X:X
[D] Network: Connecting to host public-pool.io...
[I] Network: >>> {"id":1,"method":"mining.subscribe","params":["LeafMiner/0.0.17", null]}
[I] Network: >>> {"id":2,"method":"mining.authorize","params":["XXX.esp1",""]}
[I] Network: >>> {"id":3,"method":"mining.suggest_difficulty","params":[0.000100]}
[D] Network: <<< len: 76
[I] Network: <<< [subscribe] {"id":1,"error":null,"result":[[["mining.notify","a770bbf1"]],"a770bbf1",4]}
[I] Current: New session id: a770bbf1
[D] Network: <<< len: 35
[I] Network: <<< [authorized] {"id":2,"error":null,"result":true}
[I] Network: Authorized
[D] Network: <<< len: 62
[I] Network: <<< [mining.set_difficulty] {"id":null,"method":"mining.set_difficulty","params":[0.0001]}
[I] Current: New difficulty: 0.000100000000
[D] Network: Difficulty set to: 0.0001000000
[D] Network: <<< len: 1329
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["1914621","6499fa46d4e51a1ca3417f4ffbeeaaa943c6a1560002193b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170338310d5075626c69632d506f6f6c","ffffffff0282e8f71200000000160014d5e652ee5cdcca88658c1bd8a0ad903d751e7b5b0000000000000000266a24aa21a9ed8b213665f0bf0236ae5a0517ce7360db9e33646d2bb80d1a48782c211facc00900000000",["dcb1bcf87e5b39ce2cc759ba75d94686220f16622a0001b747843578ae8f5f6a","dd556341cd62c9bd7cf957bc9a2a4e18215541b03190ef0aa742d82a35a1b09e","38e3cc1223e97532a94d4d8c212a346c5cdd4dfe916fb29aaaa39b089245b077","23b8770b0b061b23a5f864fd6d59934785a08856fc5a1a921071bdd072b594f7","6dbef30120c0c5943f8008749e1c67fce1224d6b4dea3103410a97b90221fcc4","cc0245ddaa5550ca13f31e863dea81531c00549610a161f0d04757d2e44255bf","0f179b7ca152f8f9b30270c7e8d681a3346632ee3b11ecdaeaa0aeb18442c2b7","f64b902bac1af21a112603fb36ef0663136789dc918e0e42d986db177d6a9a5c","54eebf804532dfc0d8224b605d6df7033478bfde7dc6beaa07435271f16ac696","8b4f8f1ad37fcfd1330f5f74ea987feca735be8c22bdcd9d1a0a02b288c0be89","19f3db3d4171462f38c426f8e33169f23f5e0146ea936b1b1c1031f01916106e","855e88e79d03a54b92e5624b4f9521f7e88cdd3bb506a7f923beb960e893e01f","8f8bae989a20afa05b21cd61e87675a22b8624cad0bf520cc62ef9c2d56aafbf"],"20000000","17032f14","670397c1",true]}
[I] Job: Random value: 1632096437
[I] Job: Hex value: 6147D0B5
[D] Job: Coinbase hash: 7c0d374ad943bbe7abba38ff2bf3526bda30599f5f9d7fd2251933e1cb0742c8
[D] Job: Merkle root: c41ef48ac52cf42a65253dc9d3fabace58703fe8aeb056a36834b55e475c72c3
[I] Current: Job: 1914621 ready to be mined
[D] Network: <<< len: 1263
[I] Network: <<< [mining.notify] {"id":null,"method":"mining.notify","params":["191463a","6499fa46d4e51a1ca3417f4ffbeeaaa943c6a1560002193b0000000000000000","02000000010000000000000000000000000000000000000000000000000000000000000000ffffffff170338310d5075626c69632d506f6f6c","ffffffff024c83fc1200000000160014d5e652ee5cdcca88658c1bd8a0ad903d751e7b5b0000000000000000266a24aa21a9eda11a9cb31ee3398e3f3985a33a05affc17be5cf736134caddf7bc1d161fe43c400000000",["725c6924b05a746345a535c6196fa1a07d0601303180511fa9058d8cb24d6364","6fe3c83730d81edd6f1f3307d0009f295aba86e03e36fa1b0b3c1f24bc2b7e4e","c08e331eb507c0d25f1ec52ff26956beec8f67eafba636bbfa88403594729fa3","fe12dd2f200eccf05fef92812bf585f40de300a36283f2e5c1540199b834e6f1","6ce34550578e0c29629d9fed6418c355b4ff0ad3a1447d92854288e90515193f","d7c1ce04b29f675efd4b560ddcb18aaa383963258a6017dbd4800a358e82663f","c8e8483921df67012883f123e16021f23b8e7258c6c4b79f7692dad3c81edadd","03985469846a70d62ed209a406e07fc6cf4bf2f28066c580f7a4cd4d63165c3b","a36443592b299fe15a378e133144edb2cbf3cf3a54ecd3b5f2175de33cb5420d","5dad64397a105fdd5b3aa6564a069933298e6a2816d382fd6c973ee284570749","63099f9e49e99dd8fe6ae2760f214b1d5b9cb3cbdb03050171e68b3a0a9783a6","635ebca914e4f0c9d631f9f98d43f4273a5d06c2ee89ba34849d2285a9eaef0d"],"20000000","17032f14","670397fd",false]}
[I] Job: Random value: 1632096437
[I] Job: Hex value: 6147D0B5
[D] Job: Coinbase hash: 8676648ab038f8194c148f879bcc9c31cdb93adaa9e8379a3b38722a197b0290
[D] Job: Merkle root: 3fb4441c4792907045057f5584e83ddb592de42bba8d2720bbe8885b85783a92
[I] Current: Job: 191463a ready to be mined
[D] Network: <<< len: 0
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 21.00 kH/s
[D] Current: Hashrate: 20.99 kH/s
[D] Current: Hashrate: 20.99 kH/s
[D] Miner: [0] > Hash 0.000727760514 > 0.000100000000
[I] Miner: [0] > [191463a] > 0x000dd7c7 - diff 0.000727760514
[... cut ...] 6a2c5548e3b9fa79a00fe676d10a","cd5bf862f5e2b1f7a51352d7e7aaa8bb094600b49c9d450d5f8819d7f263fc22","e8e54713c6378685780c424b557620cdfb92d1e1120875edae354c970542a6dc","aaaac5438e85f1b426d17f2dfe2b6eae0a762011ccdad23d98286cb2bf773c96"],"20000000","17032f14","67039839",false]}
[I] Job: Random value: 1632096437
[I] Job: Hex value: 6147D0B5
[D] Job: Coinbase hash: 25630e824cd41451deab90d25d9c824af4437ff52acd6ec74bc0f23737665ff5
[D] Job: Merkle root:`

Thanks again for your work, i really appreciate it!

matteocrippa, 2024-10-07:
Can be an overflow, esp8266 it's not the greatest in memory, btw when it crashes it later restarts no?
I need to check if it's not a force crash I had to trigger to make it work

g33xt4h, 2024-10-07:
> Can be an overflow, esp8266 it's not the greatest in memory, btw when it crashes it later restarts no? I need to check if it's not a force crash I had to trigger to make it work

Nope it does not recover and needs a hard reset

smartineindra, 2024-10-12:
Is there any update for this issue, I am having the same problem

gungriffonbr, 2024-12-15:
i having the same issue :(

MastaDan9, 2025-03-11:
Has anyone been able to solve the hanging problem? Let's try to figure out together how to get the esp8266 running on a permanent basis